espressif/esp-bsp #415: Updating/Replacing the existing MPU6050 driver

jeetrohan, 2024-10-22:
### Related area

    esp-bsp/components/mpu6050

### Hardware specification

None

### Is your feature request related to a problem?

Current driver for mpu6050 uses legacy i2c driver. 
Which makes it inoperable with other recent components.

### Describe the solution you'd like

I've written and tested a new driver for mpu6050 using the current i2c driver framework.
Further I've added the functionality to use the in-built motion detection feature of mpu6050.
I'm planning to update it and add more features.
Is it possible to update/replace the current driver with my contribution?

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

tore-espressif, 2024-10-22:
Hello @jeetrohan thank you for your interest.
Yes, we want to update mpu6050 to the new I2C driver. If you already have it, you can open a PR to this repository and we can help review, thanks!

jeetrohan, 2024-10-22:
> Hello @jeetrohan thank you for your interest. Yes, we want to update mpu6050 to the new I2C driver. If you already have it, you can open a PR to this repository and we can help review, thanks!

Hi @tore-espressif thanks for guidance.
I'm very new to using Github so there maybe issues with my PR (used GIthub this week for the first time, just to make this contribution).

I have simply forked the esp-bsp repo, branched it and created my files in the branch and made a PR.
Please let me know if I've done it right or do I need to redo it in some other way.

jeetrohan, 2024-10-22:
this previous close was a mistake at my end.

jeetrohan, 2025-06-06:
Closing this based on PR (BSP-568) #416 .
No longer required.

closed 2025-06-06 by jeetrohan