espressif/esp-bsp #419: BSP EXAMPLE FAIL TO BUILD DUE TO LVGL  CONFIG_LV_ATTRIBUTE_LARGE_CONST been NULL

MR-VENOM, 2024-10-25:
### Board

esp32s3eye

### Hardware Description

Plain default esp32s3 eye connected by USB

### IDE Name

VSCODE

### Operating System

linux

### Description

The example display camera is not building giving the bellow error.
```

In file included from /home/venom/disk/3rd-party/esp-bsp/examples/display_camera/managed_components/lvgl__lvgl/src/font/../../src/lv_conf_kconfig.h:15,
                 from /home/venom/disk/3rd-party/esp-bsp/examples/display_camera/managed_components/lvgl__lvgl/src/font/../../src/lv_conf_internal.h:34,
                 from /home/venom/disk/3rd-party/esp-bsp/examples/display_camera/managed_components/lvgl__lvgl/src/font/../../src/lv_init.h:16,
                 from /home/venom/disk/3rd-party/esp-bsp/examples/display_camera/managed_components/lvgl__lvgl/src/font/../../lvgl.h:21,
                 from /home/venom/disk/3rd-party/esp-bsp/examples/display_camera/managed_components/lvgl__lvgl/src/font/lv_font_montserrat_14.c:10:
/home/venom/disk/3rd-party/esp-bsp/examples/display_camera/build/config/sdkconfig.h:785:41: error: expected identifier or '(' before string constant
  785 | #define CONFIG_LV_ATTRIBUTE_LARGE_CONST ""
      |                                         ^~
/home/venom/disk/3rd-party/esp-bsp/examples/display_camera/managed_components/lvgl__lvgl/src/font/../../src/lv_conf_internal.h:1324:42: note: in expansion of macro 'CONFIG_LV_ATTRIBUTE_LARGE_CONST'
 1324 |         #define LV_ATTRIBUTE_LARGE_CONST CONFIG_LV_ATTRIBUTE_LARGE_CONST
      |                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/venom/disk/3rd-party/esp-bsp/examples/display_camera/managed_components/lvgl__lvgl/src/font/lv_font_montserrat_14.c:24:8: note: in expansion of macro 'LV_ATTRIBUTE_LARGE_CONST'
   24 | static LV_ATTRIBUTE_LARGE_CONST const uint8_t glyph_bitmap[] = {
      |        ^~~~~~~~~~~~~~~~~~~~~~~~
/home/venom/disk/3rd-party/esp-bsp/examples/display_camera/managed_components/lvgl__lvgl/src/font/lv_font_montserrat_14.c:2155:21: error: 'glyph_bitmap' undeclared here (not in a function)
 2155 |     .glyph_bitmap = glyph_bitmap,
      |                     ^~~~~~~~~~~~
[780/1032] Building C object esp-idf/lvgl__lvgl/CMakeFiles/__idf_lvgl__lvgl.dir/src/font/lv_font_montserrat_30.c.obj
ninja: build stopped: subcommand failed.

 *  The terminal process "ninja" terminated with exit code: 1. 
```

### Sketch

```cpp
/*
 * SPDX-FileCopyrightText: 2022-2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */

#include <stdio.h>
#include <string.h>
#include "sdkconfig.h"
#include "bsp/esp-bsp.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "driver/spi_master.h"
#include "esp_err.h"
#include "esp_log.h"
#include "esp_lcd_panel_io.h"
#include "esp_lcd_panel_vendor.h"
#include "esp_lcd_panel_ops.h"
#include "esp_camera.h"

static const char *TAG = "example";

void app_main(void)
{
    bsp_i2c_init();
    bsp_display_start();
    bsp_display_backlight_on(); // Set display brightness to 100%

    // Initialize the camera
    const camera_config_t camera_config = BSP_CAMERA_DEFAULT_CONFIG;
    esp_err_t err = esp_camera_init(&camera_config);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Camera Init Failed");
        return;
    }
    sensor_t *s = esp_camera_sensor_get();
    s->set_vflip(s, BSP_CAMERA_VFLIP);
    s->set_hmirror(s, BSP_CAMERA_HMIRROR);
    ESP_LOGI(TAG, "Camera Init done");

    uint32_t cam_buff_size = BSP_LCD_H_RES * BSP_LCD_V_RES * 2;
    uint8_t *cam_buff = heap_caps_malloc(cam_buff_size, MALLOC_CAP_SPIRAM);
    assert(cam_buff);

    // Create LVGL canvas for camera image
    bsp_display_lock(0);
    lv_obj_t *camera_canvas = lv_canvas_create(lv_scr_act());
    lv_canvas_set_buffer(camera_canvas, cam_buff, BSP_LCD_H_RES, BSP_LCD_V_RES, LV_COLOR_FORMAT_RGB565);
    assert(camera_canvas);
    lv_obj_center(camera_canvas);
    bsp_display_unlock();

    camera_fb_t *pic;
    while (1) {
        pic = esp_camera_fb_get();
        if (pic) {
            esp_camera_fb_return(pic);
            bsp_display_lock(0);
            memcpy(cam_buff, pic->buf, cam_buff_size);
            if (BSP_LCD_BIGENDIAN) {
                /* Swap bytes in RGB565 */
                lv_draw_sw_rgb565_swap(cam_buff, cam_buff_size);
            }
            lv_obj_invalidate(camera_canvas);
            bsp_display_unlock();
        } else {
            ESP_LOGE(TAG, "Get frame failed");
        }
        vTaskDelay(1);
    }
}
```

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

tore-espressif, 2024-10-25:
@MR-VENOM there is a bug in upstrem LVGL https://github.com/lvgl/lvgl/issues/7154

For the time being, please use LVGL v9.2.0. (You can define this in your `main/idf_component.yml`)

tore-espressif, 2024-10-29:
LVGL v9.2.1 was yanked (marked as defective) and LVGL v9.2.0 is used be default.

Thank you for the quick report!

closed 2024-10-29 by tore-espressif