espressif/esp-bsp #421: ESP-BOX-3 GT911 Touch driver not getting initialized.

cellux1002, 2024-10-26:
### Board

ESP32-S3-BOX-3

### Hardware Description

There is nothing else attached to the module, only the esp32-s3-box-3 itself.

### IDE Name

ESP-IDF

### Operating System

Windows 10

### Description

It appears that BSP cannot properly initialize the GT911 touch driver, that is in the esp-box-3. The BSP version 1.2.0~2 for the esp-box-3 is included via the esp component registery. Upon creating a simple project that displays an UI, the screen shows the UI and It works as expected, but the touch driver is not initialized and the touch does not provide any feedback. Image below shows the warning that the GT911 returns.
[image]

### Sketch

```cpp
"main.cpp"

#include <stdio.h>
#include "bsp/esp-bsp.h"
#include "bsp/touch.h"
#include "bsp/esp-box-3.h"
#include "ui/ui.h"
#include "lvgl.h"
#include "esp_log.h"

extern "C" void app_main(void){
    //i2c_master_init();
    esp_log_level_set("i2c", ESP_LOG_DEBUG);

    bsp_i2c_init(); 

    bsp_display_cfg_t cfg = {
        .lvgl_port_cfg = ESP_LVGL_PORT_INIT_CONFIG(),
        .buffer_size = BSP_LCD_H_RES * CONFIG_BSP_LCD_DRAW_BUF_HEIGHT,
        .double_buffer = 0,
        .flags = {
            .buff_dma = true,
        }
    };
    bsp_display_start_with_config(&cfg);

    bsp_display_backlight_on();

    bsp_display_lock(0);     
    lv_init();
    ui_init();
    lv_scr_load(ui_Screen1);
    
    while(true){
        lv_task_handler();
        vTaskDelay(10/portTICK_PERIOD_MS);
    }

}
```

### Other Steps to Reproduce

To reproduce this issue, create a new project from the template, install the esp-box-3 BSP from the esp component registery, and use the code provided in the sketch. This should return the same error.

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

cellux1002, 2024-10-30:
Perhaps that is true. But upon setting the RST ant INT pins, the driver returns an error and then not even the screen is initialized. There is also no response from touching the screen, which is odd.

cellux1002, 2024-10-30:
[image]
The INT pin is set to GPIO 3 while the RST pin is set to NC since it shares the pin with the LCD (GPIO 48).

If I recall, the initialization of the GT911 driver checks If the RST pin is not equal to GPIO NC before it proceeds with the initialization process.

cellux1002, 2024-11-07:
Nothing so far sadly. The only way that I have been able to properly initialize the touch driver is using the LovyanGFX library, but that requires the arduino component. Hoping this issue will be resolved soon.

espzav, 2024-11-08:
Hello, I am sorry, I don't understand, what is exactly issue? You need to set INT and RST pin and not working initialization? 
If you use BSP for ESP-BOX-3, the touch should work without any changes.

cellux1002, 2024-11-08:
I found that the touch does not work without any changes. Even in a new clean project with the BSP installed, the LCD works fine but there is no response from the touch controller. Perhaps I'm doing something wrong?

cellux1002, 2024-11-09:
[image]
I might be wrong, but looking at line 101 in "esp_lcd_touch_gt911.c" that's included in the BSP, the if statement checks to make sure that the reset pin is not set to GPIO_NUM_NC, but from my understanding it is. Could that be the reason that the condition is not met, and that it instead returns the warning at line 135?

espzav, 2024-11-11:
Hi, I tried this example now: https://github.com/espressif/esp-bsp/tree/master/examples/display_audio_photo with my `esp-box-3` and it is working properly. Touch is working. 
Compiled by: `idf.py -D SDKCONFIG_DEFAULTS=sdkconfig.bsp.esp-box-3 -p COM19 flash monitor`

The mentioned condition is for automatic setting address of touch controller. If RST or INTR not filled, then there is default address of touch controller.

cellux1002, 2024-11-11:
Thank you for the quick response, I will try to recreate your steps as soon as I get home.

I see I was mistaken with that condition, my apologies.

cellux1002, 2024-11-11:
I have retraced your steps and even though I received the same GT911 initialization error, the touch worked flawlessly in the example. After reviewing my code (code snipper in the original post), and replacing "bsp_display_start_with_config(&cfg);" with "bsp_display_start();", and removing the "lv_scr_load(ui_Screen1);" line, touch started to work. Thank you for your help, the issue seems to be resolved.

closed 2024-11-11 by cellux1002