espressif/esp-bsp #426: i2c_oled_example garbled display with LVGL 9.2

mutatrum, 2024-10-29:
### Board

ESP32-S3-WROOM-1

### Hardware Description

0.91" OLED 128x32 display

### IDE Name

vscode

### Operating System

Linux

### Description

Garbled pixels when running the example after fixing some compiler errors for LVGL 9.2. It works fine with LVGL 8.4. I initially opened https://github.com/espressif/esp-idf/issues/14784 but I think it belongs in this repo. A lot of details including screenshots are there.

### Sketch

```cpp
`idf.py create-project-from-example "espressif/esp_lvgl_port^2.3.2:i2c_oled"`

Changes I made:

Move `lv_disp_set_rotation` inside of the `lvgl_port_lock` as it triggers WDT.

Hardcoded `lv_obj_set_width` in `lvgl_demo_ui.c` to 128, as the compiler complained about incomplete `lv_display_t` interface.

Applied the following settings:

CONFIG_IDF_TARGET="esp32s3"
CONFIG_EXAMPLE_SSD1306_HEIGHT_32=y
CONFIG_LV_COLOR_DEPTH_1=y
CONFIG_LV_MEM_SIZE_KILOBYTES=32
CONFIG_LV_TXT_BREAK_CHARS=" ,.;:-_"
CONFIG_LV_USE_OBSERVER=y
CONFIG_LV_USE_SNAPSHOT=y
```
```

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

espzav, 2024-10-29:
@mutatrum Please, could you check this: https://github.com/espressif/esp-bsp/issues/398#issuecomment-2376072637 ?

mutatrum, 2024-10-29:
Unfortunately, that doesn't fix it. Most of the screen is now white, with a distinct double buffer flickering. There is some hint of things moving from left to right, so it has the correct color_map location, but it seems the format is still off.

mutatrum, 2024-11-01:
I don't seem to get it working. I tried a lot of different things, but it always looks like the incoming data in `_lvgl_port_transform_monochrome.color_map` is already garbled. I'm on an ESP32S3 and a 132.x32 display, maybe that triggers some different behaviour upstream?

espzav, 2024-11-01:
When I tried it, I had to fullclean (remove build folder, sdkconfig, managed_components and dependencies.lock) then I removed sdkconfig.defaults (it is for LVGL8). And build it with LVGL 9. Then it was working. 
We are working on update LVGL port with support `LV_COLOR_FORMAT_I1` .

mutatrum, 2024-11-01:
Even with a full clean I can't get it working. Looking forward to I1 support again, that would hopefully make it easier to get working again.

mutatrum, 2024-11-01:
Ah I misread your comment, I didn't clear out the sdkconfig.defaults. With that removed and only setting the display height to 32 in menuconfig it works.

My guess is `CONFIG_LV_COLOR_DEPTH_1=y` messes up things and has LVGL set to a half working 1 bit mode, or at least in an incompatible fashion for the ssd1306 support. I don't know if esp_port should panic when that is set, or if it can override that setting before init?

closed 2024-11-12 by mutatrum