espressif/esp-bsp #429: Unable to register custom interrupt callback for touch panel as lvgl_port_touch already does so

wreyford, 2024-11-05:
### Board

ESP32S3 16MB flash, 8 MB PSRAM, ili9488, FT5x06 custom PCB

### Hardware Description

Custom PCB

### IDE Name

VSCodue Insiders

### Operating System

Win10

### Description

I ported to LVGL9.2 from 8.4. Previously I did the portation myself, but chose to use the espressif__esp_lvgl_port.
All working very nicely.

I used to register a callback on the CPT_INT_PIN in ISR.
Now I cannot use a user defined callback, as already used by esp_lvgl_port.

### Sketch

```cpp
My old way of achieving it:

static void IRAM_ATTR ctp_gpio_isr_handler(void *arg)
{
    uint32_t gpio_num = (uint32_t)arg;
    xQueueOverwriteFromISR(lcd_touch_bkl_evt_queue, &gpio_num, NULL);
}
```
In setup code, I now tried:
```
   esp_lcd_touch_config_t tp_cfg = {
        .x_max = LCD_HEIGHT,
        .y_max = LCD_WIDTH,
        .rst_gpio_num = -1,
        .int_gpio_num = GPIO_INPUT_IO_CTP_INT, 
        .levels = {                                 //Added levels AJW 2024-11-03 12:16
            .reset = 0,
            .interrupt = 0,
        },
        .flags = {
            .swap_xy = 1,
            .mirror_x = 1,
            .mirror_y = 0,
        },
        .interrupt_callback = my_custom_lcd_touch_interrupt_cb,
    };
```
and all the usual necessary code to setup, and then the portation:
```
/* Add touch input (for selected screen) */
    ESP_LOGI(TAG, "LVGL - Add Touch");
    const lvgl_port_touch_cfg_t touch_cfg = {
        .disp = lvgl_disp,
        .handle = tp,
    };
    lvgl_touch_indev = lvgl_port_add_touch(&touch_cfg);
```
```

### Other Steps to Reproduce

In another RTOS task I wait on  lcd_touch_bkl_revent_queue and reset certain timers applicable to our software.
eg. We turn lcd backlight off after a certain time of inactivity to conserve power, and on again after touch.

I need a way to obtain touch events when they happen (only the fact that touch happened ie as with cpt_interrupt.)

How does the portation allow for this. 
I could not find any public API to achieve this. I do not want to poll for events.

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

wreyford, 2024-11-05:
I just added :
```
lv_indev_set_read_cb(lvgl_touch_indev, my_custom_lcd_touch_interrupt_cb);
```
and it worked. Didn't realize that I had to add it at the level of the lv_indev_t and not higher up.
I hope this helps someone else.

The callback defined as follows:
```
static void my_custom_lcd_touch_interrupt_cb(lv_indev_t * indev_drv, lv_indev_data_t * data)
{
    ESP_LOGW(TAG, "=================TOUCH INTERRUPT=====================");
    uint32_t gpio_num = (uint32_t)GPIO_INPUT_IO_CTP_INT;
    xQueueOverwrite(lcd_touch_bkl_evt_queue, &gpio_num);
}
```

wreyford, 2024-11-05:
Unfortunately, this breaks touch. Now I have to implement my own touch functionality.
So perhaps somebody can help me to do it the correct way.
Thx

wreyford, 2024-11-05:
So it looks like I needed this instead:
```
lv_indev_add_event_cb(lvgl_touch_indev, my_custom_lcd_touch_interrupt_cb, LV_EVENT_ALL, NULL);
```

tore-espressif, 2024-11-12:
@wreyford yes, if this `lv_indev_add_event_cb(lvgl_touch_indev, my_custom_lcd_touch_interrupt_cb, LV_EVENT_ALL, NULL);` works for you, then this would be the best recommended option.

Is there anything else we could do in this issue?

closed 2024-12-11 by tore-espressif