espressif/esp-bsp #443: LVGL(managed components) causing a crash after a random time (IDFGH-14133)

akashpraan, 2024-11-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.4

### Espressif SoC revision.

ESP32 S3 Wroom 1U (16MB Flash)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I expect the display to get updated properly everytime and not result in a crash after a random number of hours 

### What is the actual behavior?

The ESP resets after a random number of hours 

I want to figure out why there is this failure and why it is affecting my system and getting the ESP to restart after a random amount of hours.
The last time I observed it was after 12 hours of leaving the system ON and on 12 or 15th button press this issue was seen.

lvgl/lvgl: “^8.3.0”
esp_lvgl_port: “^1”
These have been added in dependencies in idf_component.yml

### Steps to reproduce.

I am using a Black&White 0.96inch OLED Display to display status of Fan Mode and the sensor values

1. There is a button(or via an MQTT Command), the user can change the Fan Mode and this is reflected on the OLED and after 5 seconds(using timer) Sensor data is display(the sensor data is refreshed every 2 minutes)
2. On Some Custom PCB (4/40 for now), which doing extended testing we observe that after a certain while the ESP restarts citing that there has been an attempt to do “memcpy in ROM”
3. Now, I tried reproducing the issue by pressing the button at close intervals (like 30 times a minute) and I didn’t observe a crash for 3-4 minutes straight and then I stopped, currently my system has been running normally for 3.5Hours with me repeating this 30 times a minute thing for 2 minutes every hour.

For one instance I got the error back trace from the ESP such as
```c
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core  0 register dump:
PC      : 0x420168ea  PS      : 0x00060b30  A0      : 0x820169db  A1      : 0x3fcd57e0
--- 0x420168ea: get_prop_core at C:/Akash/Firmwares/Hive/Hive_E_ESPIDF_514/Hive_E/managed_components/lvgl__lvgl/src/core/lv_obj_style.c:613
A2      : 0x3fcaf568  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00000002  A7      : 0xb33fffff  A8      : 0x820168bd  A9      : 0x3fcd57c0
A10     : 0x00000001  A11     : 0x00588f0e  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x00060023  A15     : 0x00000003  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff
--- 0x40056f5c: memcpy in ROM
0x40056f72: memcpy in ROM
[... 153 lines cut ...]
        }
    }

    lv_label_set_text(label_value, buf);        // Update the text of the value label.

    // Get screen dimensions
    lv_coord_t screen_width = lv_obj_get_width(scr);
    lv_coord_t screen_height = lv_obj_get_height(scr);

    // Set a fixed y-coordinate for vertical alignment (center of the screen)
    lv_coord_t label_y = screen_height / 2 - 9;  // Static vertical alignment 
    lv_coord_t value_y = screen_height / 2 - 15;  // Static vertical alignment 
    lv_coord_t label_x = 12;  // Set a fixed x-coordinate for the label (left-aligned), Fixed padding from the left

    lv_coord_t value_x = screen_width - 58; // Set a fixed x-coordinate for the value (right-aligned), Fixed padding from the right 
    lv_obj_align(label_main, LV_ALIGN_TOP_LEFT, label_x, label_y);      // Align the main label to the left, with fixed y-coordinate
    lv_obj_align(label_value, LV_ALIGN_TOP_LEFT, value_x, value_y);     // Align the value label to the right, with fixed y-coordinate

    #if LCD_DISPLAY_DEBUG == ON
    ESP_LOGI("DisplayData", "Exiting display_data function");
    ESP_LOGI("DisplayData", " ");
    #endif
}

```

igrr, 2024-11-27:
@akashpraan Your program is crashing because it tries to dereference a NULL pointer. Please check how to interpret panic output in this section of the docs: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/fatal-errors.html#loadprohibited-storeprohibited

For your case, I would recommend enabling GDB Stub and then waiting for the program to crash. Once the program has crashed it will enter the GDB session and you can inspect the variables to understand what happened.

On a brief look, it does seem like the issue happens due to the fact that you are manipulating LVGL state in display_data function from your task at the same time as LVGL is performing a refresh in its own task. Typically you need to hold the muted while modifying UI state outside of LVGL task.

akashpraan, 2024-11-27:
Hello @igrr I am using LVGL for the first time, what mutex are you talking about?

Sorry to bother with this questions, I tried finding the answer but couldn't.

tore-espressif, 2024-11-27:
> I am using LVGL for the first time, what mutex are you talking about?

LVGL is not thread safe, you must call this function 

https://github.com/espressif/esp-bsp/blob/48936a65c8bd5bad95de0582b5ad15e0f35558c3/components/esp_lvgl_port/include/esp_lvgl_port.h#L104

Before you call any LVGL function (LVGL functions start with `lv_*`). After you are done, call 
https://github.com/espressif/esp-bsp/blob/48936a65c8bd5bad95de0582b5ad15e0f35558c3/components/esp_lvgl_port/include/esp_lvgl_port.h#L110

These two functions take/release mutex that protects LVGL internal data while you are modifying the UI

So it should look like this
``` c
...
lvgl_port_lock(1000); // Take LVGL lock (implemented as mutex)
lv_*();               // Do what you need with your UI
lvgl_port_unlock();   // Release the lock
...
```

igrr, 2024-11-27:
Here is the related part of the docs, for reference: https://github.com/espressif/esp-bsp/blob/48936a65c8bd5bad95de0582b5ad15e0f35558c3/components/esp_lvgl_port/README.md#lvgl-api-usage

tore-espressif, 2024-12-11:
@akashpraan We'll close this issue, feel free to reopen if you have any more questions

closed 2024-12-11 by tore-espressif