espressif/esp-bsp #445: LVGL failed to build ThorVG

MR-VENOM, 2024-11-29:
### Board

esp32s3eye

### Hardware Description

Plain old esp32s3eye module

### IDE Name

VSCODE

### Operating System

linux

### Description

I am trying to run the lottie example in LVGL and for that DOCS syay to enable THORvg in config, and also enable lottie, But the program failed to build by giving internal compiler error on thorVG. 

### Sketch

```cpp
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"

#include "lv_demos.h"
#include "bsp/esp-bsp.h"
#include "lvgl.h"
#include "lv_examples.h"

static char *TAG = "app_main";

extern "C" void app_main(void)
{
    bsp_display_start();
    bsp_display_backlight_on();

    ESP_LOGI(TAG, "Display LVGL demo");
    bsp_display_lock(0);
    extern const uint8_t lv_example_lottie_approve[];
    extern const size_t lv_example_lottie_approve_size;

    lv_obj_t *lottie = lv_lottie_create(lv_screen_active());
    lv_lottie_set_src_data(lottie, lv_example_lottie_approve, lv_example_lottie_approve_size);

    static uint8_t buf[240 * 240 * 4];
    lv_lottie_set_buffer(lottie, 240, 240, buf);

    lv_obj_center(lottie);
    bsp_display_unlock();
}
```

### Other Steps to Reproduce

extra files :

[idf_component.yml.txt](https://github.com/user-attachments/files/17958718/idf_component.yml.txt)
[sdkconfig.txt](https://github.com/user-attachments/files/17958715/sdkconfig.txt)

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

espressif2022, 2024-11-29:
Hello，can you provide the compilation error log?

igrr, 2024-11-29:
Linking https://github.com/jcmvbkbc/gcc-xtensa/issues/14 — adding `-fno-if-conversion` to compilation flags could help.

The issue will be finally fixed in toolchain based on GCC 14, which is used in IDF v5.4 or later.

MR-VENOM, 2024-11-29:
Will try on IDF 5.4

MR-VENOM, 2024-11-30:
> Linking [jcmvbkbc/gcc-xtensa#14](https://github.com/jcmvbkbc/gcc-xtensa/issues/14) — adding `-fno-if-conversion` to compilation flags could help.
> 
> The issue will be finally fixed in toolchain based on GCC 14, which is used in IDF v5.4 or later.

Now the code builds but it crashed on loading

###LOG

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40376402  PS      : 0x00060e30  A0      : 0x80376641  A1      : 0x3fcb2250  
0x40376402: heap_caps_malloc_base at /home/venom/esp/v5.2.1/esp-idf/components/heap/heap_caps.c:155

A2      : 0x00001800  A3      : 0x000000bc  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00001000  A8      : 0x3fca72b0  A9      : 0x01ffffff  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3fca5470  
A14     : 0x3fca3db4  A15     : 0x00000002  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000d8  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  
0x400570e8: memset in ROM
0x400570f3: memset in ROM

Backtrace: 0x403763ff:0x3fcb2250 0x4037663e:0x3fcb2280 0x40376798:0x3fcb22b0 0x40391f55:0x3fcb22d0 0x4204183d:0x3fcb22f0 0x42041905:0x3fcb2310 0x42041d89:0x3fcb2340 0x4204329d:0x3fcb7100 0x42040e99:0x3fcb71f0 0x4204496e:0x3fcb7240 0x4203f52a:0x3fcb7260 0x4204112d:0x3fcb7280 0x42034eeb:0x3fcb72b0 0x4203520a:0x3fcb7390 0x42034f15:0x3fcb73e0 0x4203520a:0x3fcb74c0 0x42034f15:0x3fcb7510 0x42034fa9:0x3fcb75f0 0x4205c79d:0x3fcb76d0 0x420b3705:0x3fcb7710 0x42052673:0x3fcb7730 0x4205269d:0x3fcb7750 0x42045a03:0x3fcb7770 0x420477ee:0x3fcb77a0 0x42047949:0x3fcb77d0 0x42015995:0x3fcb77f0
.....0x403763ff: heap_caps_malloc_base at /home/venom/esp/v5.2.1/esp-idf/components/heap/heap_caps.c:151 (discriminator 1)
0x4037663e: heap_caps_realloc_base at /home/venom/esp/v5.2.1/esp-idf/components/heap/heap_caps.c:409
0x40376798: heap_caps_realloc_default at /home/venom/esp/v5.2.1/esp-idf/components/heap/heap_caps.c:269
0x40391f55: realloc at /home/venom/esp/v5.2.1/esp-idf/components/newlib/heap.c:34
0x4204183d: _genSpan(SwRleData*, SwSpan const*, unsigned long) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRle.cpp:317
0x42041905: _horizLine(RleWorker&, long, long, long, long) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRle.cpp:380
0x42041d89: rleRender(SwRleData*, SwOutline const*, SwBBox const&, bool) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRle.cpp:426 (discriminator 2)
 (inlined by) rleRender(SwRleData*, SwOutline const*, SwBBox const&, bool) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRle.cpp:994 (discriminator 2)
0x4204329d: shapeGenStrokeRle(SwShape*, tvg::RenderShape const*, tvg::Matrix const*, SwBBox const&, SwBBox&, SwMpool*, unsigned int) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwShape.cpp:617
0x42040e99: SwShapeTask::run(unsigned int) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRenderer.cpp:170
0x4204496e: tvg::TaskSchedulerImpl::request(tvg::Task*) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgTaskScheduler.cpp:189
 (inlined by) tvg::TaskScheduler::request(tvg::Task*) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgTaskScheduler.cpp:217
0x4203f52a: tvg::SwRenderer::prepareCommon(SwTask*, tvg::RenderTransform const*, tvg::Array<void*> const&, unsigned char, tvg::RenderUpdateFlag) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRenderer.cpp:765
0x4204112d: tvg::SwRenderer::prepareCommon(SwTask*, tvg::RenderTransform const*, tvg::Array<void*> const&, unsigned char, tvg::RenderUpdateFlag) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRenderer.cpp:723
 (inlined by) tvg::SwRenderer::prepare(tvg::RenderShape const&, void*, tvg::RenderTransform const*, tvg::Array<void*>&, unsigned char, tvg::RenderUpdateFlag, bool) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRenderer.cpp:809
0x42034eeb: tvg::Shape::Impl::update(tvg::RenderMethod*, tvg::RenderTransform const*, tvg::Array<void*>&, unsigned char, tvg::RenderUpdateFlag, bool) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgShape.h:110
 (inlined by) tvg::Paint::Impl::update(tvg::RenderMethod*, tvg::RenderTransform const*, tvg::Array<void*>&, unsigned char, tvg::RenderUpdateFlag, bool) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgPaint.cpp:293
0x4203520a: tvg::Scene::Impl::update(tvg::RenderMethod*, tvg::RenderTransform const*, tvg::Array<void*>&, unsigned char, tvg::RenderUpdateFlag, bool) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgScene.h:125
0x42034f15: tvg::Paint::Impl::update(tvg::RenderMethod*, tvg::RenderTransform const*, tvg::Array<void*>&, unsigned char, tvg::RenderUpdateFlag, bool) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lv [... cut ...]
{
    bsp_display_start();
    bsp_display_backlight_on();

    load_spiffs();

    ESP_LOGI(TAG, "Display LVGL demo");
    bsp_display_lock(0);

    lv_obj_t *lottie = lv_lottie_create(lv_screen_active());
    lv_lottie_set_src_file(lottie, "/spiffs/lv_example_lottie_approve.json");

    uint8_t *buf = (uint8_t *)heap_caps_malloc(240 * 240 * 4,MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
    lv_lottie_set_buffer(lottie, 240, 240, buf);

    lv_obj_center(lottie);
    bsp_display_unlock();

    
}
```

MR-VENOM, 2024-11-30:
Solved it ..  Need to allocate LVGL memory on PSRAM istead of DMAR.

> > Linking [jcmvbkbc/gcc-xtensa#14](https://github.com/jcmvbkbc/gcc-xtensa/issues/14) — adding `-fno-if-conversion` to compilation flags could help.
> > The issue will be finally fixed in toolchain based on GCC 14, which is used in IDF v5.4 or later.
> 
> @igrr Now the code builds but it crashed on loading
> 
> ###LOG
> 
> ```
> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> 
> Core  1 register dump:
> PC      : 0x40376402  PS      : 0x00060e30  A0      : 0x80376641  A1      : 0x3fcb2250  
> 0x40376402: heap_caps_malloc_base at /home/venom/esp/v5.2.1/esp-idf/components/heap/heap_caps.c:155
> 
> A2      : 0x00001800  A3      : 0x000000bc  A4      : 0x00000000  A5      : 0x00000000  
> A6      : 0x00000000  A7      : 0x00001000  A8      : 0x3fca72b0  A9      : 0x01ffffff  
> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3fca5470  
> A14     : 0x3fca3db4  A15     : 0x00000002  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
> EXCVADDR: 0x000000d8  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  
> 0x400570e8: memset in ROM
> 0x400570f3: memset in ROM
> 
> 
> 
> Backtrace: 0x403763ff:0x3fcb2250 0x4037663e:0x3fcb2280 0x40376798:0x3fcb22b0 0x40391f55:0x3fcb22d0 0x4204183d:0x3fcb22f0 0x42041905:0x3fcb2310 0x42041d89:0x3fcb2340 0x4204329d:0x3fcb7100 0x42040e99:0x3fcb71f0 0x4204496e:0x3fcb7240 0x4203f52a:0x3fcb7260 0x4204112d:0x3fcb7280 0x42034eeb:0x3fcb72b0 0x4203520a:0x3fcb7390 0x42034f15:0x3fcb73e0 0x4203520a:0x3fcb74c0 0x42034f15:0x3fcb7510 0x42034fa9:0x3fcb75f0 0x4205c79d:0x3fcb76d0 0x420b3705:0x3fcb7710 0x42052673:0x3fcb7730 0x4205269d:0x3fcb7750 0x42045a03:0x3fcb7770 0x420477ee:0x3fcb77a0 0x42047949:0x3fcb77d0 0x42015995:0x3fcb77f0
> .....0x403763ff: heap_caps_malloc_base at /home/venom/esp/v5.2.1/esp-idf/components/heap/heap_caps.c:151 (discriminator 1)
> 0x4037663e: heap_caps_realloc_base at /home/venom/esp/v5.2.1/esp-idf/components/heap/heap_caps.c:409
> 0x40376798: heap_caps_realloc_default at /home/venom/esp/v5.2.1/esp-idf/components/heap/heap_caps.c:269
> 0x40391f55: realloc at /home/venom/esp/v5.2.1/esp-idf/components/newlib/heap.c:34
> 0x4204183d: _genSpan(SwRleData*, SwSpan const*, unsigned long) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRle.cpp:317
> 0x42041905: _horizLine(RleWorker&, long, long, long, long) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRle.cpp:380
> 0x42041d89: rleRender(SwRleData*, SwOutline const*, SwBBox const&, bool) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRle.cpp:426 (discriminator 2)
>  (inlined by) rleRender(SwRleData*, SwOutline const*, SwBBox const&, bool) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRle.cpp:994 (discriminator 2)
> 0x4204329d: shapeGenStrokeRle(SwShape*, tvg::RenderShape const*, tvg::Matrix const*, SwBBox const&, SwBBox&, SwMpool*, unsigned int) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwShape.cpp:617
> 0x42040e99: SwShapeTask::run(unsigned int) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRenderer.cpp:170
> 0x4204496e: tvg::TaskSchedulerImpl::request(tvg::Task*) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgTaskScheduler.cpp:189
>  (inlined by) tvg::TaskScheduler::request(tvg::Task*) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgTaskScheduler.cpp:217
> 0x4203f52a: tvg::SwRenderer::prepareCommon(SwTask*, tvg::RenderTransform const*, tvg::Array<void*> const&, unsigned char, tvg::RenderUpdateFlag) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRenderer.cpp:765
> 0x4204112d: tvg::SwRenderer::prepareCommon(SwTask*, tvg::RenderTransform const*, tvg::Array<void*> const&, unsigned char, tvg::RenderUpdateFlag) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRenderer.cpp:723
>  (inlined by) tvg::SwRenderer::prepare(tvg::RenderShape const&, void*, tvg::RenderTransform const*, tvg::Array<void*>&, unsigned char, tvg::RenderUpdateFlag, bool) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRenderer.cpp:809
> 0x42034eeb: tvg::Shape::Impl::update(tvg::RenderMethod*, tvg::RenderTransform const*, tvg::Array<void*>&, unsigned char, tvg::RenderUpdateFlag, bool) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgShape.h:110
>  (inlined by) tvg::Paint::Impl::update(tvg::RenderMethod*, tvg::RenderTransform const*, tvg::Array<void*>&, unsigned char, tvg::RenderUpdateFlag, bool) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgPaint.cpp:293
> 0x4203520a: tvg::Scene::Impl::update(tvg::RenderMethod*, tvg::RenderTransform const*, tvg::Array<void*>&, unsigned char, tvg::RenderUpdateFlag, bool) at /home/venom/disk/custom-esp32-examples/lvgl-lottie/managed_components/lvgl__lvgl/src/libs/thorvg/tvgScene.h:125
[... cut ...]
    bsp_display_backlight_on();

    load_spiffs();

    ESP_LOGI(TAG, "Display LVGL demo");

    bsp_display_lock(0);
    lv_obj_t *lottie = lv_lottie_create(lv_screen_active());
    lv_lottie_set_src_file(lottie, "/spiffs/lv_example_lottie_approve.json");
    uint8_t *buf = (uint8_t *)heap_caps_malloc(LOTTIE_H * LOTTIE_W * 4,MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
    // uint8_t buf[LOTTIE_H * LOTTIE_W * 4];
    lv_lottie_set_buffer(lottie, LOTTIE_H, LOTTIE_W, buf);
    lv_obj_center(lottie);
    bsp_display_unlock();
}

```

closed 2024-11-30 by MR-VENOM