espressif/esp-bsp #449: SquareLine自定义ESP开发板

zeandon, 2024-12-03:
我在网上看到了这个链接：https://www.bilibili.com/opus/721643227044642853。里面提到了可以通过修改esp-bsp 仓库中的示例：custom_waveshare_7inch，设计自己的开发板模板。
在查看了这个文件之后，我发现这块7英寸屏幕的接口是I2C和并行数据线，所以想问一下能否对这个文件进行修改，实现驱动一块驱动芯片为ILI9341的240x320分辨率的TFT触摸屏幕（触摸和数据都使用SPI进行传输），具体引脚定义如下：
[image]
如果可以实现，需要修改 示例custom_waveshare_7inch 中的哪些部分？
同时，这份2022年的文章（https://www.bilibili.com/opus/721643227044642853）中提到的代码好像与现在的文件中的代码有所差异，官方有更新关于SquareLine自定义ESP32开发板的相关内容吗？
打扰了，希望能收到回复，非常感谢！

zeandon, 2024-12-06:
我按照上面提到的文章对ws_7inch.c、ws_7inch.h、idf_component.yml、mainfest,json进行了修改：
在ws_7inch.c中，将include的头文件改为
`#include "esp_timer.h"
#include "driver/gpio.h"
#include "driver/ledc.h"
#include "esp_err.h"
#include "esp_log.h"
#include "esp_check.h"
#include "esp_lcd_panel_io.h"
#include "esp_lcd_panel_vendor.h"
#include "esp_lcd_panel_ops.h"

#include "bsp/ws_7inch.h"
#include "esp_lcd_ili9341.h"
#include "esp_lcd_touch_stmpe610.h"
#include "esp_lvgl_port.h"
#include "bsp_err_check.h"
#include "bsp/display.h"
#include "bsp/touch.h"`
将bsp_i2c_init函数修改为bsp_spi_init函数：
`
esp_err_t bsp_spi_init(void)
{
    ESP_LOGI(TAG, "Initialize SPI bus");
    spi_bus_config_t buscfg = {
        .sclk_io_num = BSP_T_CLK,
        .mosi_io_num = BSP_T_MOSI,
        .miso_io_num = BSP_T_MISO,
        .quadwp_io_num = -1, // Quad SPI LCD driver is not yet supported
        .quadhd_io_num = -1, // Quad SPI LCD driver is not yet supported
        .max_transfer_sz = BSP_LCD_H_RES * 80 * sizeof(uint16_t), // transfer 80 lines of pixels (assume pixel is RGB565) at most in one SPI transaction
    };
    ESP_ERROR_CHECK(spi_bus_initialize(T_HOST, &buscfg, SPI_DMA_CH_AUTO)); // Enable the DMA feature

    return ESP_OK;
}`
将bsp_display_new函数中关于初始化i2c和8080的代码修改为：
`
esp_err_t bsp_display_new(const bsp_display_config_t *config, esp_lcd_panel_handle_t *ret_panel, esp_lcd_panel_io_handle_t *ret_io)
{
    esp_err_t ret = ESP_OK;

    ESP_LOGI(TAG, "Initialize SPI bus");
    spi_bus_config_t buscfg = {
        .sclk_io_num = BSP_LCD_SCK,
        .mosi_io_num = BSP_LCD_MOSI,
        .miso_io_num = BSP_LCD_MISO,
        .quadwp_io_num = -1, // Quad SPI LCD driver is not yet supported
        .quadhd_io_num = -1, // Quad SPI LCD driver is not yet supported
        .max_transfer_sz = BSP_LCD_H_RES * 80 * sizeof(uint16_t), // transfer 80 lines of pixels (assume pixel is RGB565) at most in one SPI transaction
    };
    ESP_ERROR_CHECK(spi_bus_initialize(LCD_HOST, &buscfg, SPI_DMA_CH_AUTO)); // Enable the DMA feature

    ESP_LOGI(TAG, "Install panel IO");
    esp_lcd_panel_io_handle_t io_handle = NULL;
    esp_lcd_panel_io_spi_config_t io_config = {
        .dc_gpio_num = BSP_LCD_DC,
        .cs_gpio_num = BSP_LCD_CS,
        .pclk_hz = BSP_LCD_PIXEL_CLOCK_HZ,
        .lcd_cmd_bits = LCD_CMD_BITS,
        .lcd_param_bits = LCD_PARAM_BITS,
        .spi_mode = 0,
        .trans_queue_depth = 10,
    };
    // Attach the LCD to the SPI bus
    ESP_ERROR_CHECK(esp_lcd_new_panel_io_spi((esp_lcd_spi_bus_handle_t)LCD_HOST, &io_config, &io_handle));

    esp_lcd_panel_handle_t panel_handle = NULL;
    esp_lcd_panel_dev_config_t panel_config = {
        .reset_gpio_num = EXAMPLE_PIN_NUM_RST,
[... 166 lines cut ...]
I (111) boot: End of partition table
I (116) esp_image: segment 0: paddr=00020020 vaddr=3c060020 size=168d8h ( 92376) map
I (138) esp_image: segment 1: paddr=00036900 vaddr=3fc9a100 size=0340ch ( 13324) load
I (141) esp_image: segment 2: paddr=00039d14 vaddr=40378000 size=06304h ( 25348) load
I (149) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=5ad64h (372068) map
I (209) esp_image: segment 4: paddr=0009ad8c vaddr=4037e304 size=0bd50h ( 48464) load
I (226) boot: Loaded app from partition at offset 0x20000
I (226) boot: Disabling RNG early entropy source...
E (238) octal_psram: PSRAM ID read error: 0x00000000, PSRAM chip not found or not supported, or wrong PSRAM line mode
E cpu_start: Failed to init external RAM, needed for external .bss segment
﻿
abort() was called at PC 0x403796a1 on core 0
--- 0x403796a1: call_start_cpu0 at /home/zando/esp/esp-idf/components/esp_system/port/cpu_start.c:578
﻿
﻿
﻿
Backtrace: 0x40379d12:0x3fceb220 0x40382341:0x3fceb240 0x40387c9d:0x3fceb260 0x403796a1:0x3fceb2d0 0x403cca4c:0x3fceb340 0x403cce6d:0x3fceb380 0x403c89a5:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
--- 0x40379d12: panic_abort at /home/zando/esp/esp-idf/components/esp_system/panic.c:463
0x40382341: esp_system_abort at /home/zando/esp/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x40387c9d: abort at /home/zando/esp/esp-idf/components/newlib/abort.c:38
0x403796a1: call_start_cpu0 at /home/zando/esp/esp-idf/components/esp_system/port/cpu_start.c:578
0x40045c01: ets_run_flash_bootloader in ROM
0x40043ab6: main in ROM
0x40034c45: .stack_ok in ROM`
请问这是因为对bsp文件修改出错而导致的问题吗？

espzav, 2025-03-13:
HI @zeandon,
it seems, that issue is not related to LCD. Check this error:
```
E (238) octal_psram: PSRAM ID read error: 0x00000000, PSRAM chip not found or not supported, or wrong PSRAM line mode
```
You should change PSRAM mode or disable it (depends on your board): `SPIRAM_MODE_QUAD=y` or `PSRAM=n`

espzav, 2025-09-10:
Hi @zeandon, any update?