espressif/esp-bsp #451: LVGL Lottie draw buffer cannot allocate from external memory esp32s3

MR-VENOM, 2024-12-04:
### Board

esp32s3eye

### Hardware Description

Using the esp32s3 eye module,

### IDE Name

VSCODE

### Operating System

linux

### Description

The lv_lottie_set_buffer is not able to set more than 180 *180 buffer even though psram is enabled and set to allocate by malloc() and have 8mb psram i am using it with esp-sr for speech recognition the speech recogntion works finr but Lottie on LVGL does not work

[sdkconfig.txt](https://github.com/user-attachments/files/18003334/sdkconfig.txt)

### Sketch

```cpp
lvgl_port_cfg_t lvgl_port_cfg = {
        .task_priority = CONFIG_BSP_DISPLAY_LVGL_TASK_PRIORITY,
        .task_stack = 6144,
        .task_affinity = 1,
        .task_max_sleep_ms = CONFIG_BSP_DISPLAY_LVGL_MAX_SLEEP,
        .timer_period_ms = CONFIG_BSP_DISPLAY_LVGL_TICK,
    };
    ESP_ERROR_CHECK(lvgl_port_init(&lvgl_port_cfg));

    bsp_display_config_t bsp_disp_cfg = {
        .max_transfer_sz = BSP_LCD_DRAW_BUFF_SIZE * sizeof(uint16_t),
    };
    ESP_ERROR_CHECK(bsp_display_new(&bsp_disp_cfg, &this->panel_handle, &this->io_handle));

    esp_lcd_panel_disp_on_off(this->panel_handle, true);

    lvgl_port_display_cfg_t disp_cfg = {
        .io_handle = this->io_handle,
        .panel_handle = this->panel_handle,
        .buffer_size = BSP_LCD_DRAW_BUFF_SIZE * 4,
        .double_buffer = 1,
        .hres = BSP_LCD_H_RES,
        .vres = BSP_LCD_V_RES,
        .monochrome = false,
        /* Rotation values must be same as used in esp_lcd for initial settings of the screen */
        .rotation = {
            .swap_xy = false,
            .mirror_x = false,
            .mirror_y = false,
        },
        .flags = {
            .buff_dma = false,
            .buff_spiram = true,
#if LVGL_VERSION_MAJOR >= 9
            .swap_bytes = (BSP_LCD_BIGENDIAN ? true : false),
#endif
        }
    };
    
    this->disp = lvgl_port_add_disp(&disp_cfg);
    assert(this->disp);

    bsp_display_backlight_on();

    this->gfx_buffer = (uint8_t *)heap_caps_malloc(240 * 240 * 4, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
    assert(this->gfx_buffer != nullptr);
    bsp_display_lock(0);
    this->lottie_obj = lv_lottie_create(lv_screen_active());
    bsp_display_unlock();
    
    
    bsp_display_lock(0);
    lv_lottie_set_src_file(this->lottie_obj, path.c_str());
    lv_lottie_set_buffer(this->lottie_obj, 10, 10, this->gfx_buffer);
    lv_obj_center(this->lottie_obj);
    bsp_display_unlock();
```

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

espressif2022, 2024-12-04:
Hello, you can try to increase stack size  `.task_stack = 6144`, thorvg needs quite larger stack.

MR-VENOM, 2024-12-04:
STill the isuue is not solved.

I changes the stack size into  ```.task_stack = 6144 * 5 ```

### LOG

```

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x403835c1  PS      : 0x00060e33  A0      : 0x80388dda  A1      : 0x3fca2840  
--- 0x403835c1: spinlock_acquire at /home/nikhi/esp/v5.2.3/esp-idf/components/esp_hw_support/include/spinlock.h:103
 (inlined by) xPortEnterCriticalTimeout at /home/nikhi/esp/v5.2.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:474

A2      : 0x00000031  A3      : 0xffffffff  A4      : 0x3fca75b4  A5      : 0x00060e23  
A6      : 0x00000000  A7      : 0x0000cdcd  A8      : 0x00060e20  A9      : 0x3fca2840  
A10     : 0x00060e23  A11     : 0x00000000  A12     : 0x00060e20  A13     : 0x3fca2d4c  
A14     : 0x00000001  A15     : 0x000000e7  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000031  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  
--- 0x400570e8: memset in ROM
0x400570f3: memset in ROM

Backtrace: 0x403835be:0x3fca2840 0x40388dd7:0x3fca2870 0x40388e16:0x3fca2890 0x4037a71d:0x3fca28b0 0x4037a74a:0x3fca28d0 0x4037a4ec:0x3fca2900 0x4038a631:0x3fca2920 0x420384a5:0x3fca2940 0x4203867e:0x3fca2960 0x420387d9:0x3fca2980 0x420397f7:0x3fca7720 0x42037629:0x3fca7750 0x4203b036:0x3fca77a0 0x42037c37:0x3fca77c0 0x42037caf:0x3fca77e0 0x4202dffe:0x3fca7810 0x4202e465:0x3fca7850 0x4202e2aa:0x3fca7900 0x4202e486:0x3fca7950 0x4202e2aa:0x3fca7a00 0x4202e486:0x3fca7a50 0x4202e583:0x3fca7b00 0x4202e4a7:0x3fca7b80 0x4205d6a5:0x3fca7c30 0x4205d93d:0x3fca7c70 0x4205d95f:0x3fca7ca0 0x420231bf:0x3fca7cd0 0x4204b171:0x3fca7d00 0x42008efa:0x3fca7d30 0x42008dd0:0x3fca7d50 0x421220cf:0x3fca7d90 0x403833d1:0x3fca7dc0
--- Error: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
--- Waiting for the device to reconnect.
--- 0x403835be: xt_utils_get_raw_core_id at /home/nikhi/esp/v5.2.3/esp-idf/components/xtensa/include/xt_utils.h:51
 (inlined by) spinlock_acquire at /home/nikhi/esp/v5.2.3/esp-idf/components/esp_hw_support/include/spinlock.h:87
 (inlined by) xPortEnterCriticalTimeout at /home/nikhi/esp/v5.2.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:474
0x40388dd7: xPortEnterCriticalTimeoutSafe at /home/nikhi/esp/v5.2.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:581
 (inlined by) vPortEnterCriticalSafe at /home/nikhi/esp/v5.2.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:588
 (inlined by) multi_heap_internal_lock at /home/nikhi/esp/v5.2.3/esp-idf/components/heap/multi_heap.c:165
0x40388e16: multi_heap_malloc_impl at /home/nikhi/esp/v5.2.3/esp-idf/components/heap/multi_heap.c:209
0x4037a71d: heap_caps_malloc_base at /home/nikhi/esp/v5.2.3/esp-idf/components/heap/heap_caps_base.c:131
0x4037a74a: heap_caps_realloc_base at /home/nikhi/esp/v5.2.3/esp-idf/components/heap/heap_caps_base.c:159
0x4037a4ec: heap_caps_realloc_default at /home/nikhi/esp/v5.2.3/esp-idf/components/heap/heap_caps.c:151
0x4038a631: realloc at /home/nikhi/esp/v5.2.3/esp-idf/components/newlib/heap.c:34
0x420384a5: _genSpan(SwRleData*, SwSpan const*, unsigned long) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRle.cpp:317
0x4203867e: _sweep(RleWorker&) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRle.cpp:434 (discriminator 1)
0x420387d9: rleRender(SwRleData*, SwOutline const*, SwBBox const&, bool) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRle.cpp:994
0x420397f7: shapeGenStrokeRle(SwShape*, tvg::RenderShape const*, tvg::Matrix const*, SwBBox const&, SwBBox&, SwMpool*, unsigned int) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwShape.cpp:617
0x42037629: SwShapeTask::run(unsigned int) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRenderer.cpp:170
0x4203b036: tvg::TaskSchedulerImpl::request(tvg::Task*) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgTaskScheduler.cpp:189
 (inlined by) tvg::TaskScheduler::request(tvg::Task*) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgTaskScheduler.cpp:217
0x42037c37: tvg::SwRenderer::prepareCommon(SwTask*, tvg::RenderTransform const*, tvg::Array<void*> const&, unsigned char, tvg::RenderUpdateFlag) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRenderer.cpp:765
0x42037caf: tvg::SwRenderer::prepare(tvg::RenderShape const&, void*, tvg::RenderTransform const*, tvg::Array<void*>&, unsigned char, tvg::RenderUpdateFlag, bool) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRenderer.cpp:809
0x4202dffe: tvg::Shape::Impl::update(tvg::RenderMethod*, tvg::RenderTransform const*, tvg::Array<void*>&, unsigned char, tvg::RenderUpdateFlag, bool) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgShape.h:110
0x4202e465: tvg::Paint::Impl::update(tvg::RenderMethod*, tvg::RenderTransform const*, tvg::Array<void*>&, unsigned char, tvg::RenderUpdateFlag, bool) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgPaint.cpp:293 (discriminator 2)
[... cut ...]
0x403833d1: vPortTaskWrapper at /home/nikhi/esp/v5.2.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: c5856b30c

--- Error: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
--- Waiting for the device to reconnect.
--- Error: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
--- Waiting for the device to reconnect.
--- Error: device reports readiness to read but 

```

MR-VENOM, 2024-12-04:
Let me break down the components of my project:

ESP-DL with examples from ESP-WHO.
ESP-SR for English speech recognition.
Wi-Fi for internet connectivity.
ESP-HTTP for server communication.
LVGL for Lottie animations.
Everything else works as expected, but the Lottie files only function properly if their dimensions are 100x100 or smaller.

MR-VENOM, 2024-12-04:
And one more thing i notice is when my wifi is running my spi are not able to transfer image to the display

```

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x568
load:0x403c9700,len:0x4
load:0x403c9704,len:0xb5c
load:0x403cc700,len:0x2d10
entry 0x403c98f0
I (35) cpu_start: Multicore app
I (36) octal_psram: vendor id    : 0x0d (AP)
I (36) octal_psram: dev id       : 0x02 (generation 3)
I (39) octal_psram: density      : 0x03 (64 Mbit)
I (44) octal_psram: good-die     : 0x01 (Pass)
I (49) octal_psram: Latency      : 0x01 (Fixed)
I (54) octal_psram: VCC          : 0x01 (3V)
I (59) octal_psram: SRF          : 0x01 (Fast Refresh)
I (65) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (71) octal_psram: BurstLen     : 0x01 (32 Byte)
I (76) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (82) octal_psram: DriveStrength: 0x00 (1/1)
I (88) MSPI Timing: PSRAM timing tuning index: 5
I (93) esp_psram: Found 8MB PSRAM device
I (97) esp_psram: Speed: 80MHz
I (214) mmu_psram: Instructions copied and mapped to SPIRAM
I (357) mmu_psram: Read only data copied and mapped to SPIRAM
I (518) esp_psram: SPI SRAM memory test OK
I (526) cpu_start: Pro cpu start user code
I (526) cpu_start: cpu freq: 240000000 Hz
I (527) cpu_start: Application information:
I (527) cpu_start: Project name:     ai-module
I (527) cpu_start: App version:      9116a8c-dirty
I (528) cpu_start: Compile time:     Dec  4 2024 13:53:14
I (529) cpu_start: ELF file SHA256:  e1d339b76...
I (529) cpu_start: ESP-IDF:          v5.2.3-dirty
I (529) cpu_start: Min chip rev:     v0.0
I (530) cpu_start: Max chip rev:     v0.99 
I (530) cpu_start: Chip rev:         v0.1
I (531) heap_init: Initializing. RAM available for dynamic allocation:
I (531) heap_init: At 3FCAA890 len 0003EE80 (251 KiB): RAM
I (532) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (533) esp_psram: Adding pool of 4082K of PSRAM memory to heap allocator
I (533) spi_flash: detected chip: generic
I (534) spi_flash: flash io: qio
W (535) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (542) sleep: Configure to isolate all GPIO pins in sleep state
I (543) sleep: Enable automatic switching of GPIO sleep configuration
sion e7ae62f
I (544) main_task: Started on CPU0
I (545) esp_psram: Reserving pool of 16K of internal memory for DMA/internal allocations
I (546) main_task: Calling app_main()
W (546) S3-EYE: Warning: Long filenames on SD card are disabled in menuconfig!
I (546) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (547) gpio: GPIO[38]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (548) gpio: GPIO[40]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (606) pp: pp rom version: e7ae62f
I (606) net80211: net80211 rom version: e7ae62f
I (608) wifi:wifi driver task: 3fcbb264, prio:23, stack:6656, core=0
I (611) wifi:wifi firmware version: 82f54d2
I (612) wifi:wifi certification version: v7.0
I (612) wifi:config NVS flash: enabled
I (612) wifi:config nano formating: disabled
I (612) wifi:Init data frame dynamic rx buffer num: 128
I (613) wifi:Init static rx mgmt buffer num: 5
I (613) wifi:Init management short buffer num: 32
I (614) wifi:Init static tx buffer num: 32
I (614) wifi:Init tx cache buffer num: 32
I (615) wifi:Init static tx FG buffer num: 2
[... 80 lines cut ...]
I (3299) S3-EYE: Setting LCD backlight: 100%
E (3307) lcd_panel.io.spi: panel_io_spi_tx_color(390): spi transmit (queue) color failed
E (3307) lcd_panel.st7789: panel_st7789_draw_bitmap(224): io tx color failed
I (3335) WIFI-PROV: WIFI DISCONNECTED
I (3335) WIFI-PROV: WAIT FOR WIFI CONNECT
I (4336) WIFI-PROV: WIFI DISCONNECTED
I (4336) WIFI-PROV: WAIT FOR WIFI CONNECT
I (5336) WIFI-PROV: WIFI DISCONNECTED
I (5336) WIFI-PROV: WAIT FOR WIFI CONNECT
I (5690) WIFI-PROV: connect to the AP fail
I (5694) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (5694) wifi:state: init -> auth (b0)
I (5770) wifi:state: auth -> assoc (0)
I (5819) wifi:state: assoc -> run (10)
I (5837) wifi:connected with test, aid = 5, channel 11, BW20, bssid = 9e:d6:44:db:71:94
I (5837) wifi:security: WPA2-PSK, phy: bgn, rssi: -45
I (5838) wifi:pm start, type: 1

I (5838) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (5839) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (5856) wifi:<ba-add>idx:0 (ifx:0, 9e:d6:44:db:71:94), tid:0, ssn:0, winSize:64
I (5872) wifi:dp: 2, bi: 102400, li: 4, scale listen interval from 307200 us to 409600 us
I (5872) wifi:AP's beacon interval = 102400 us, DTIM period = 2

```

espressif2022, 2024-12-04:
> 4038a631: realloc at /home/nikhi/esp/v5.2.3/esp-idf/components/newlib/heap.c:34

If you disable other module, it can load the size large than 100*100？Maybe there's no memory.
we can run [http, sr, and lottie(240*240) together](https://www.bilibili.com/video/BV1MezzYaEfP/?spm_id_from=333.337.search-card.all.click).

MR-VENOM, 2024-12-04:
> If you disable other module, it can load the size large than 100*100？Maybe there's no memory. we can run [http, sr, and lottie(240*240) [...]

yes that my exact point . i have 8MB psram , isisnt that enough , On my startup only the wifi is enabled . My startups procedure is like this

```

  bsp_sdcard_mount();
  WifiProv *wifi = new WifiProv();
  std::string scan,ssid = "test",pass ="12345678";
  wifi->scan(&scan);
  wifi->connect(ssid, pass);
  
  Gfx *gfx = new Gfx();

  gfx->lottie("/sdcard/lottie/wifi.json");

  wifi->wait_for_wifi();

```

it crashes at :

``` gfx->lottie("/sdcard/lottie/wifi.json"); ```

But the   ```   this->gfx_buffer = (uint8_t *)heap_caps_malloc(240 * 240 * 4, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT); ``` is able to allocate from PSRAM

Only problem happens is when i call  ``` lv_lottie_set_buffer(this->lottie_obj, 240, 240, this->gfx_buffer);```

MR-VENOM, 2024-12-04:
> we can run [http, sr, and lottie(240*240) together](https://www.bilibili.com/video/BV1MezzYaEfP/?spm_id_from=333.337.search-card.all.click).

This looks amazing is this open source ? if so can you share the files ?

espressif2022, 2024-12-04:
> > we can run [http, sr, and lottie(240*240) together](https://www.bilibili.com/video/BV1MezzYaEfP/?spm_id_from=333.337.search-card.all.click).
> 
> This looks amazing is this open source ? if so can you share the files ?

Yeah,  it should be tomorrow, I think you can refer to the lottie function, it use the extra thorvg.

espressif2022, 2024-12-04:
Yes, this is not the reason. Can you share the json file for me to try?

MR-VENOM, 2024-12-04:
@espressif2022  [this one](https://github.com/user-attachments/files/18005818/wifi.json)

MR-VENOM, 2024-12-05:
Hi @espressif2022  any update on this ?

espressif2022, 2024-12-05:
> Hi @espressif2022 any update on this ?

Indeed, I can't play any one of your json, even if the size is set to less than 100*100

MR-VENOM, 2024-12-05:
> > Hi @espressif2022 any update on this ?
> 
> Indeed, I can't play any one of your json, even if the size is set to less than 100*100

I am able to play the file if my project contain only the lottie file. But whenever i add components to it . it breaks

espressif2022, 2024-12-09:
I see no error output when running, which is a bit difficult. Even the version is ThorVG 0.14.9.

MR-VENOM, 2024-12-11:
> I see no error output when running, which is a bit difficult. Even the version is ThorVG 0.14.9.

So what is the sollution to this ? :cry: 

> Yeah, it should be tomorrow, I think you can refer to the lottie function, it uses the extra thorvg.

Any update on this ? so that i can check what you guys are doing different

espressif2022, 2024-12-11:
> > I see no error output when running, which is a bit difficult. Even the version is ThorVG 0.14.9.
> 
> So what is the sollution to this ? 😢
> 
> > Yeah, it should be tomorrow, I think you can refer to the lottie function, it uses the extra thorvg.
> 
> Any update on this ? so that i can check what you guys are doing different

I use this [component](https://components.espressif.com/components/espressif2022/lottie_player/versions/1.0.0), the API is the same with LVGL.

MR-VENOM, 2025-01-07:
@espressif2022  Can you provide any solution to this ? do you have ANY lottie files that worked for you ?

PawnMa, 2025-01-23:
I encountered the same issue. When I use the built-in thorvg, as long as the size of my Lottie drawing exceeds 100x100, it will cause problems. Otherwise, drawing four at the same time is fine. I adjusted the task stack area, but it didn't help. May I ask if your problem has been resolved? （Translate with GPT）

klaatu2a, 2025-03-10:
> [@espressif2022](https://github.com/espressif2022) Can you provide any solution to this ? do you have ANY lottie files that worked for you ?

Did you resolve this issue?  
Do you have the ESP32 running on LVGL 9.2.2?

PawnMa, 2025-03-11:
I resolved the issue, which was related to the task stack size configuration. However, it's important to note that since I used the lvgl_port library, the application and the lvgl main loop are not running in the same thread. Therefore, it is necessary to set the stack size for the thread in which the application runs. Once I set an appropriate size, the stack overflow issue was resolved. @klaatu2a @MR-VENOM @

MR-VENOM, 2025-03-11:
@klaatu2a @PawnMa After days of debugging, I finally got it to work and display a 240x240 lottie. The main issue was that LVGL doesn’t allocate one large chunk of memory at a time — instead, it requests lots of small chunks, usually under 100 bytes. In ESP-IDF, to make SPIRAM usable with the standard C library (malloc and friends), you need to enable the SPIRAM_USE_MALLOC config. But that brings its own set of problems.

There’s another config called SPIRAM_MALLOC_ALWAYSINTERNAL. This setting defines a size threshold — any allocation below this size will always be placed in internal memory. In my case, this was set to 1024 bytes, and since LVGL allocates in small chunks, all of them were ending up in internal RAM instead of external RAM. The fix was to set SPIRAM_MALLOC_ALWAYSINTERNAL to 0, so everything, regardless of size, could go into external RAM.

On top of that, to render a 240x240 lottie, you need a minimum stack size of 24,576 bytes.

closed 2025-03-11 by MR-VENOM