espressif/esp-bsp #453: No free memmory

mike88macedon, 2024-12-08:
### Board

Esp32P4EvBoard

### Hardware Description

Mipi dsi ek70097 display

### IDE Name

VSCose

### Operating System

Windows 10

### Description

Problem with bsp_init() when initializing  ek70097 mipi display with exception no free memmory.

### Sketch

```cpp
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "nvs_flash.h"
#include "nvs.h"
#include "esp_log.h"
#include "esp_err.h"
#include "esp_check.h"
#include "esp_memory_utils.h"
#include "lvgl.h"
#include "bsp/esp-bsp.h"
#include "bsp/display.h"
#include "bsp_board_extra.h"
#include "lv_demos.h"

void app_main(void)
{
    bsp_display_cfg_t cfg = {
        .lvgl_port_cfg = ESP_LVGL_PORT_INIT_CONFIG(),
        .buffer_size = BSP_LCD_DRAW_BUFF_SIZE,
        .double_buffer = BSP_LCD_DRAW_BUFF_DOUBLE,
        .flags = {
            .buff_dma = true,
            .buff_spiram = false,
            .sw_rotate = false,
        }
    };
    bsp_display_start_with_config(&cfg);
    bsp_display_backlight_on();

    bsp_display_lock(0);

    lv_demo_music();
    // lv_demo_benchmark();
    // lv_demo_widgets();

    bsp_display_unlock();
}
```

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

espzav, 2025-01-03:
Hello @mike88macedon, please did you try the `display_lvgl_demos` example without any changes? And did you use `sdkconfig.bsp.esp32_p4_function_ev_board` with your build?

Indastri, 2025-03-03:
Since there was no response from the user - closing the issue. If you encounter any problems please reopen this or create a new issue.

Cheers!

closed 2025-03-03 by Indastri