espressif/esp-bsp #456: Doc: Readme does not contain reference to ESP32-S3-BOX-3

georgik, 2024-12-13:
### Board

ESP32-S3-BOX-3

### Hardware Description

ESP32-S3-BOX-3

### IDE Name

CLion

### Operating System

macOS

### Description

The REAMDE file https://github.com/espressif/esp-bsp?tab=readme-ov-file#supported-boards does not contain link to ESP32-S3-BOX-3, there is just link to older variant ESP32-S3-BOX.

Please, update the README, so that it reflects boards which are available on the market.

### Sketch

```cpp
ESP32-S3-BOX-3
```

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

espzav, 2024-12-13:
Here it is: 
[image]

georgik, 2024-12-13:
@espzav Please, update the README, so that boards are sorted eaither by order. And preference should have boards which are available on the market.

closed 2025-01-06 by tore-espressif