espressif/esp-bsp #462: Memory Leak in lvgl_port_remove_disp Related to draw_ctx

huhaifan, 2024-12-22:
### Board

Custom board

### Hardware Description

Custom board

### IDE Name

VSCode

### Operating System

Windows11

### Description

## Environment:

LVGL Version: 8.4.0
esp_lvgl_port Version: 2.4.3
## Description:
I have encountered a memory leak when calling lvgl_port_remove_disp after lvgl_port_add_disp. The issue seems to originate from the draw_ctx not being properly deinitialized and freed.

1. Call lvgl_port_add_disp to add a display.
1. Call lvgl_port_remove_disp to remove the display.
1. Observe the memory leak related to draw_ctx.

## Proposed Fix:
To resolve this issue, I suggest modifying the lvgl_port_remove_disp function to include the following code:
```c
if (disp_drv->draw_ctx) {
    disp_drv->draw_ctx_deinit(disp_drv, disp_drv->draw_ctx);
    lv_mem_free(disp_drv->draw_ctx);
    disp_drv->draw_ctx = NULL;
}
```

### Sketch

```cpp
lvgl_port_add_disp() then lvgl_port_remove_disp()
```

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

espzav, 2025-02-13:
Hi @huhaifan, I am not sure, but it should be free inside LVGL. If this is issue, it is on LVGL side or LVGL documentation. I didn't see anything ybout it in official documentation. 
@kisvegabor Please, could you look at it and confirm my opinion?

kisvegabor, 2025-02-24:
@espzav I've just check that it's really a memory leak in LVGL. We don't free `draw_ctx` anywhere. 🙁 

We don't have any planned release for v8.4 patches, so probably it's faster to fix on your end.

PR #516 (2025-02-26): fix(lvgl_port): Fixed memory leak in LVGL8 in display removing
# ESP-BSP Pull Request checklist

- [ ] Version of modified component bumped
- [ ] CI passing

# Change description
- Fixed memory leak in LVGL8 in display removing
Closes #462
closed 2025-02-26 by espzav