espressif/esp-bsp #464: Error setting LV_COLOR_DEPTH_1 when trying to use LVGL v9

abouatef, 2024-12-30:
### Board

ESP32 Wroom

### Hardware Description

SSD1306

### IDE Name

ESP-IDF extension for VSCode

### Operating System

windows 11

### Description

Hi,

Trying to run the i1c_oled example I get the following error when using LVGL v9.2.2
#error please disable LV_COLOR_DEPTH_1 for using monochromatic screen

 And if I remove the LV_COLOR_DEPTH_1 config then the display uses 16-bit by default and doesn't work properly

### Sketch

```cpp
i1c_oled example
```

### I have checked existing issues, README.md and ESP32 Forum

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

espzav, 2025-01-03:
Hello @abouatef, the LVGL port not support `LV_COLOR_DEPTH_1` option with LVGL9. Instead of this, you can change color format to `LV_COLOR_FORMAT_I1` here: https://github.com/espressif/esp-bsp/blob/056eb6959fd092a66ac035b8fb6d76bbbfb3799e/components/esp_lvgl_port/examples/i2c_oled/main/i2c_oled_example_main.c#L129
The example should work with both color formats. 

Please, don't forget remove these files and folder before build (if you compiled for LVGL8 before):
- build/
- managed_components/
- dependencies.lock
- sdkconfig
- sdkconfig.defaults

suda-morris, 2025-01-07:
Another example FYI: https://github.com/espressif/esp-idf/blob/master/examples/peripherals/lcd/i2c_oled/main/i2c_oled_example_main.c#L199

espzav, 2025-03-06:
@abouatef Closing this issue. Feel free to reopen it, if you need.

closed 2025-03-06 by espzav