espressif/esp-bsp #465: Add Support for 10" Touch Display waveshare/esp_lcd_jd9365_10_1

wapel, 2025-01-01:
### Related area

Configure the Touch display jd9365 like ek79007 and ili9881c

### Hardware specification

esp32-p4-function-ev-board

### Is your feature request related to a problem?

I don't get the esp_lcd_jd9365 running with lvl on my esp-p4-function-ev-board.

### Describe the solution you'd like

The ability to configure the Waveshare Display jd9365via the SDK Configuration:
Selection LCD display Board Support Package(ESP32-P4) --> Display --> Select LCD type

LCD 1280x800 - ili9881c (default)
LCD 1024x600 - ek79007
HDMI - lt8912b

### Describe alternatives you've considered

Failed to add it manually

### Additional context

Component is already existing:
https://components.espressif.com/components/waveshare/esp_lcd_jd9365_10_1/versions/0.1.0

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

espzav, 2025-01-03:
Hello @wapel BSP repository support only Espressif and M5Stack boards with components, which are delivered with boards. I am sorry. You should contact owner of the component for help.

espzav, 2025-03-06:
Closing this issue for inactivity. Feel free to reopen it.

closed 2025-03-06 by espzav