espressif/esp-bsp #466: Support for anothers sub boards

ManelCS, 2025-01-02:
### Related area

display

### Hardware specification

esp32-s3-lcd-ev-board

### Is your feature request related to a problem?

I'm not sure if you can use other sub boards than those already supported.

### Describe the solution you'd like

I would like to know if it is possible to use different sub boards in the esp32-s3-lcd-ev-board kit and if there could be an easy way to configure it.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

espzav, 2025-01-03:
Hello @ManelCS, in BSP are supported sub-boards only with RGB screens (480 x 480 and 800 x 480). Which board do you want to use it? Other boards can be simply configure with some LCD component from component registry.

ManelCS, 2025-01-03:
I would just like to know if these possible options can be used:
-	TFT control using 8080 16 lines
-	TFT control using 8080 8 lines 
-	TFT control using serial SPI port 8 lines
-	TFT control using two 8080 8 lines TFT’s 
-	TFT control using two SPI controlled TFT’s 

And how to use them

espzav, 2025-01-03:
This is not supported in BSP now. We can add it into our plan. All sub boards are mentioned in readme: https://github.com/espressif/esp-bsp/blob/master/bsp/esp32_s3_lcd_ev_board/README.md with drivers.

- TFT 8080 (ST7796S) - LCD driver component is here: https://components.espressif.com/components/espressif/esp_lcd_st7796/versions/1.3.1
- SPI (ST7789V) - LCD driver is included inside IDF

I have got these sub boards now. 
@Lzw655 please do you have mentioned SUB boards?

ManelCS, 2025-01-03:
And is it possible to use the previous options in the esp32s3 wroom 1 module?

espzav, 2025-01-03:
Yes, the drivers are compatible with all ESP32S3 modules.

Lzw655, 2025-01-06:
> This is not supported in BSP now. We can add it into our plan. All sub boards are mentioned in readme: https://github.com/espressif/esp-bsp/blob/master/bsp/esp32_s3_lcd_ev_board/README.md with [...]

I'm sorry, the other sub-boards have not been mass-produced.

espzav, 2025-01-06:
> I'm sorry, the other sub-boards have not been mass-produced.

@Lzw655 Thank you for info. 
@ManelCS We cannot support it in BSP, if this is not available for customers.

ManelCS, 2025-01-14:
Is it possible to use sub board 2 480x480 but only use 8 data lines and not 16?

espzav, 2025-02-17:
> What should I do to use the SPI driver (ST7789V), you say it is already included in the IDF? [@espzav](https://github.com/espzav)?

Here is the example: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/spi_lcd_touch

ManelCS, 2025-02-26:
> > What should I do to use the SPI driver (ST7789V), you say it is already included in the IDF? [@espzav](https://github.com/espzav)?
> 
> Here is the example: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/spi_lcd_touch

This example uses the GC9A01 or ILI9341 drivers, what should I do if I want to use the ST7789V?

espzav, 2025-02-26:
You should change [this](https://github.com/espressif/esp-idf/blob/0461e2ff88369c3da0d4caced31e8488f53376cd/examples/peripherals/lcd/spi_lcd_touch/main/spi_lcd_touch_example_main.c#L213) line to `esp_lcd_new_panel_st7789`.

espzav, 2025-03-13:
@ManelCS Was your issue solved? Can we close this issue?

ManelCS, 2025-03-13:
> [@ManelCS](https://github.com/ManelCS) Was your issue solved? Can we close this issue?

Yes thanks!

ManelCS, 2025-04-08:
> > What should I do to use the SPI driver (ST7789V), you say it is already included in the IDF? [@espzav](https://github.com/espzav)?
> 
> Here is the example: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/spi_lcd_touch

What would I have to change to use 3-wire API (without D/C pin, 9-bit data)

espzav, 2025-04-08:
You can set [here](https://github.com/espressif/esp-idf/blob/a25e7ab59ed197817d4a78e139220b2707481f67/examples/peripherals/lcd/spi_lcd_touch/main/spi_lcd_touch_example_main.c#L196C10-L196C22) `.dc_gpio_num = GPIO_NUM_NC,` 
Command and param bits are set [here](https://github.com/espressif/esp-idf/blob/a25e7ab59ed197817d4a78e139220b2707481f67/examples/peripherals/lcd/spi_lcd_touch/main/spi_lcd_touch_example_main.c#L199)

ManelCS, 2025-04-08:
> You can set [here](https://github.com/espressif/esp-idf/blob/a25e7ab59ed197817d4a78e139220b2707481f67/examples/peripherals/lcd/spi_lcd_touch/main/spi_lcd_touch_example_main.c#L196C10-L196C22) [...]

What should I put in cmd/param_bits?

espzav, 2025-04-08:
It depends on your display. If you need 9-bits, you can put `9` to cmd or param. What display controller do you have? Isn't the driver for your controller in [Component Registry](https://components.espressif.com/)?

ManelCS, 2025-04-08:
> It depends on your display. If you need 9-bits, you can put `9` to cmd or param. What display controller do you have? Isn't the driver for your controller in [Component Registry](https://components.espressif.com/)?

I am using the MPU Interface Mode 3-line 9 bit serial I/F Ⅱ of the ST7789VI controller

espzav, 2025-04-08:
I am trying to find documentation (I found [this](https://www.orientdisplay.com/wp-content/uploads/2020/07/ST7789VI_SPEC_V1.3.pdf)) And there is not 3-SPI mode with 9bits, if I understand right. I found page 102 and there is only 8bits. Could you help me to show, which chapter is about your solution please?

ManelCS, 2025-04-08:
> I am trying to find documentation (I found [this](https://www.orientdisplay.com/wp-content/uploads/2020/07/ST7789VI_SPEC_V1.3.pdf)) And there is not 3-SPI mode with 9bits, if I understand right. I [...]

I use the penultimate mode on page 33.

[image]

espzav, 2025-04-08:
> What would I have to change to use 3-wire API (without D/C pin, 9-bit data)?
I looked into [LCD driver code](https://github.com/espressif/esp-idf/blob/a25e7ab59ed197817d4a78e139220b2707481f67/components/esp_lcd/spi/esp_lcd_panel_io_spi.c) and you should set `sio_mode` [here](https://github.com/espressif/esp-idf/blob/a25e7ab59ed197817d4a78e139220b2707481f67/components/esp_lcd/include/esp_lcd_io_spi.h#L40C22-L40C30) for 3-wire usage too. 

About 9-bits, I am not sure, how it is working. I have never used this mode. It seems, that both (cmd/param) should be 9.

ManelCS, 2025-04-09:
Amb aquest codi he aconseguit mostrar el primer frame a la pantalla, pero desptres la pantalla no s'actualitza.

#include <stdio.h>
#include <unistd.h>
#include <sys/lock.h>
#include <sys/param.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "esp_log.h"
#include "sdkconfig.h"
#include "esp_rom_sys.h"
#include "esp_timer.h"
#include "esp_err.h"
#include "esp_log.h"

#include "hal/gpio_ll.h"
#include "soc/gpio_struct.h"
#include "soc/io_mux_reg.h"

#include "lvgl.h"

#include "esp_lcd_panel_io.h"
#include "esp_lcd_panel_vendor.h"
#include "esp_lcd_panel_ops.h"
#include "driver/spi_master.h"

// Define pin numbers for the display
#define PIN_NUM_SCLK  12
#define PIN_NUM_MOSI  11
#define PIN_NUM_CS    10
#define PIN_NUM_RST   1
#define PIN_NUM_BL    41

// Define LCD dimensions
#define LCD_WIDTH     240
#define LCD_HEIGHT    320

static const char *TAG = "ST7789_LVGL";

// External function to display demo UI
extern void example_lvgl_demo_ui(lv_disp_t *disp);

static inline void spi_delay() {
    //esp_rom_delay_us(1);
}

// Function to send a 9-bit data or command to the display via SPI
static void spi_send_9bit(bool is_data, uint8_t byte) {
    uint16_t word = (is_data ? 0x100 : 0x000) | byte;

    // Optimized bit-banging
    GPIO.out_w1tc = (1 << PIN_NUM_CS);  // Set CS low

    for (int i = 8; i >= 0; i--) {
        // Clock low
        GPIO.out_w1tc = (1 << PIN_NUM_SCLK);

        if ((word >> i) & 0x01)
            GPIO.out_w1ts = (1 << PIN_NUM_MOSI);  // Set MOSI high if bit is 1
        else
            GPIO.out_w1tc = (1 << PIN_NUM_MOSI);  // Set MOSI low if bit is 0

        // Clock high
        GPIO.out_w1ts = (1 << PIN_NUM_SCLK);
    }

    GPIO.out_w1ts = (1 << PIN_NUM_CS);  // Set CS high
}

// Function to send a command to the LCD
static void lcd_send_cmd(uint8_t cmd) { spi_send_9bit(false, cmd); }
// Function to send data to the LCD
static void lcd_send_data(uint8_t data) { spi_send_9bit(true, data); }

// Function to reset the LCD display
static void lcd_reset() {
    gpio_set_level(PIN_NUM_RST, 0);  // Set reset pin low
    vTaskDelay(pdMS_TO_TICKS(100));  // Wait for 100ms
    gpio_set_level(PIN_NUM_RST, 1);  // Set reset pin high
    vTaskDelay(pdMS_TO_TICKS(100));  // Wait for 100ms
}

// Function to initialize the ST7789 LCD display
static void st7789_init() {
    lcd_reset();
    lcd_send_cmd(0x01); vTaskDelay(pdMS_TO_TICKS(150));  // Software reset
    lcd_send_cmd(0x11); vTaskDelay(pdMS_TO_TICKS(120));  // Sleep out
    lcd_send_cmd(0x36); lcd_send_data(0x00);  // Memory data access control
    lcd_send_cmd(0x3A); lcd_send_data(0x55);  // Pixel format
    lcd_send_cmd(0x21);  // Inversion on
    lcd_send_cmd(0x29);  // Display on
    vTaskDelay(pdMS_TO_TICKS(50));  // Wait for 50ms
}

// Function to set the window for drawing on the display
static void lcd_set_window(uint16_t x0, uint16_t y0, uint16_t x1, uint16_t y1) {
    lcd_send_cmd(0x2A);  // Set column address
    lcd_send_data(x0 >> 8); lcd_send_data(x0 & 0xFF);
    lcd_send_data(x1 >> 8); lcd_send_data(x1 & 0xFF);
    lcd_send_cmd(0x2B);  // Set row address
    lcd_send_data(y0 >> 8); lcd_send_data(y0 & 0xFF);
    lcd_send_data(y1 >> 8); lcd_send_data(y1 & 0xFF);
    lcd_send_cmd(0x2C);  // Write to memory
}

// Function to flush the screen buffer to the display
void my_flush_cb(lv_disp_drv_t *disp_drv, const lv_area_t *area, lv_color_t *color_p) {

    int32_t w = area->x2 - area->x1 + 1;  // Calculate width of the area
    int32_t h = area->y2 - area->y1 + 1;  // Calculate height of the area

    lcd_set_window(area->x1, area->y1, area->x2, area->y2);  // Set window to draw

    // Send the buffer contents to the screen
    for (int y = 0; y < h; y++) {
        for (int x = 0; x < w; x++) {
            uint16_t color = lv_color_to16(*color_p);  // Convert LVGL color to RGB565
            lcd_send_data(color >> 8);  // Send high byte
            lcd_send_data(color & 0xFF);  // Send low byte
            color_p++;  // Move to the next pixel
        }
    }

    // Signal that the flush is complete
    lv_disp_flush_ready(disp_drv);  // Mark the display as ready for next update
}

// ========== app_main ==============
// Main application entry point
void app_main(void) {
    gpio_config_t io_conf = {
        .mode = GPIO_MODE_OUTPUT,
        .pin_bit_mask = (1ULL << PIN_NUM_SCLK) |
                        (1ULL << PIN_NUM_MOSI) |
                        (1ULL << PIN_NUM_CS)   |
                        (1ULL << PIN_NUM_RST)  |
                        (1ULL << PIN_NUM_BL)   // Configure GPIO pins
    };
    gpio_config(&io_conf);
    gpio_set_level(PIN_NUM_CS, 1);  // Set CS high
    gpio_set_level(PIN_NUM_SCLK, 1);  // Set SCLK high
    gpio_set_level(PIN_NUM_BL, 1);  // Set backlight on

    ESP_LOGI(TAG, "Initializing ST7789...");
    st7789_init();  // Initialize the display

    lv_init();  // Initialize LVGL library

    // Memory for the screen buffers
    static lv_color_t buf1[LCD_WIDTH * 40];  // First buffer with 40 lines
    static lv_color_t buf2[LCD_WIDTH * 40];  // Second buffer for double buffering

    // Create the display driver
    static lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);  // Initialize display driver structure

    disp_drv.hor_res = LCD_WIDTH;  // Set horizontal resolution
    disp_drv.ver_res = LCD_HEIGHT;  // Set vertical resolution
    disp_drv.flush_cb = my_flush_cb;  // Set flush callback for screen updates
    static lv_disp_draw_buf_t draw_buf;
    lv_disp_draw_buf_init(&draw_buf, buf1, buf2, LCD_WIDTH * 40);  // Initialize the draw buffer
    disp_drv.draw_buf = &draw_buf;

    // Register the display driver
    lv_disp_drv_register(&disp_drv);

    example_lvgl_demo_ui(NULL);  // Run the LVGL demo UI

    // Main loop
    while (1) {
        lv_timer_handler();  // Handle LVGL tasks
        vTaskDelay(pdMS_TO_TICKS(20));  // Delay to allow for updates
    }
}

espzav, 2025-04-10:
@ManelCS, I checked it with our driver team and the 3-wire 9-bit is not supported in `esp_lcd` yet.

espzav, 2025-04-14:
We have new issue for this here #561, I am closing this issue.

closed 2025-04-14 by espzav