espressif/esp-bsp #475: LVGL9 crashing with esp32-S3 on esp-idf 4.4.8 (IDFGH-14391)

ankit-thealchemist, 2025-01-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf v4.4.8

### Espressif SoC revision.

Chip is ESP32-S3 (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32 s3 devkit C1

### Power Supply used.

USB

### What is the expected behavior?

 I want to use the ssd1309 with lvgl. I have modified the example tjpgd and connect the lvgl 8.3.11, which is working good. 
 How ever when I tired with lvgl 9.0.0, its crasing (logs and screenshot attached).
 
 How ever, if I place a delay of 10 milli sec in the file esp_lvgl_port.c after creating the task `lvgl_port_task` it works flawlessly. 
 Seems like before the task has been created, device is reached at a point to access the location which has to be created by the above task. 
 
 
[image]

[image]

### What is the actual behavior?

Device should not crashed on on using the lvgl 9 version. 

### Steps to reproduce.

1. Step checkout to branch v4.4.8
2. Step Add the dependency esp32_lvgl_port
3. Step add the lvgl 9.0.0
4. build flash and monitor
...

### Debug Logs.

```plain
Erst:0x15 (USB_UART_CHIP_RESET),boot:0x2f (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048de8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x16cc
load:0x403c9700,len:0xbd4
load:0x403cc700,len:0x2fb0
entry 0x403c9954
I (18) boot: ESP-IDF v4.4.8-dirty 2nd stage bootloader
I (18) boot: compile time 15:55:00
I (18) boot: Multicore bootloader
I (21) boot: chip revision: v0.1
I (24) boot.esp32s3: Boot SPI Speed : 80MHz
I (29) boot.esp32s3: SPI Mode       : DIO
I (34) boot.esp32s3: SPI Flash Size : 2MB
I (39) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
[[0;32mI (55) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (62) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (70) boot:  2 factory          factory app      00 00 00010000 00100000
I (77) boot: End of partition table
I (82) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=106e8h ( 67304) map
I (102) esp_image: segment 1: paddr=00020710 vaddr=3fc92c80 size=02194h (  8596) load
I (104) esp_image: segment 2: paddr=000228ac vaddr=40374000 size=0d76ch ( 55148) load
I (120) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=4dfach (319404) map
I (178) esp_image: segment 4: paddr=0007dfd4 vaddr=4038176c size=0150ch (  5388) load
I (185) boot: Loaded app from partition at offset 0x10000
I (185) boot: Disabling RNG early entropy source...
I (198) cpu_start: Multicore app
I (198) cpu_start: Pro cpu up.
I (198) cpu_start: Starting app cpu, entry point is 0x40375248
0x40375248: call_start_cpu1 at /home/ankit/sdk-soft/esp-idf/components/esp_system/port/cpu_start.c:151

I (0) cpu_start: App cpu up.
I (216) cpu_start: Pro cpu start user code
I (216) cpu_start: cpu freq: 160000000
I (216) cpu_start: Application information:
I (219) cpu_start: Project name:     lcd_tjpgd
I (224) cpu_start: App version:      e858734b-dirty
I (229) cpu_start: Compile time:     Jan  9 2025 15:56:44
I (235) cpu_start: ELF file SHA256:  91afdae8b273f900...
I (241) cpu_start: ESP-IDF:          v4.4.8-dirty
I (247) cpu_start: Min chip rev:     v0.0
I (251) cpu_start: Max chip rev:     v0.99 
I (256) cpu_start: Chip rev:         v0.1
I (261) heap_init: Initializing. RAM available for dynamic allocation:
I (268) heap_init: At 3FCA59F0 len 00043D20 (271 KiB): D/IRAM
I (275) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
I (281) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (288) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (295) spi_flash: detected chip: generic
I (299) spi_flash: flash io: dio
W (303) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (316) sleep: Configure to isolate all GPIO pins in sleep state
I (323) sleep: Enable automatic switching of GPIO sleep configuration
I (330) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (350) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (360) gpio: GPIO[46]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (360) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (490) main: Initialize LVGL
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Setting breakpoint at 0x4201b531 and retur0x4201b531: search_suitable_block at /home/profile/practice_code/esp32_cpp/ESP32-Examples/esp-idf/ssd1309_lvgl/managed_components/lvgl__lvgl/src/stdlib/builtin/lv_tlsf.c:563
```

igrr, 2025-01-09:
> search_suitable_block at /home/profile/practice_code/esp32_cpp/ESP32-Examples/esp-idf/ssd1309_lvgl/managed_components/lvgl__lvgl/src/stdlib/builtin/lv_tlsf.c:563

Could you please attach your sdkconfig? Looks like you are using LVGL's builtin heap implementation.

ankit-thealchemist, 2025-01-09:
> > search_suitable_block at /home/profile/practice_code/esp32_cpp/ESP32-Examples/esp-idf/ssd1309_lvgl/managed_components/lvgl__lvgl/src/stdlib/builtin/lv_tlsf.c:563
> 
> Could you please attach your sdkconfig? Looks like you are using LVGL's builtin heap implementation.

[sdkconfig.txt](https://github.com/user-attachments/files/18362688/sdkconfig.txt)

tore-espressif, 2025-01-16:
@ankit-thealchemist could you please check if https://github.com/espressif/esp-bsp/pull/468 fixes your problem?

closed 2025-01-16 by tore-espressif