espressif/esp-bsp #479: 希望可以推出gc9d01的驱动库

NingZiXi, 2025-01-14:
### Related area

显示屏驱动

### Hardware specification

esp32s3

### Is your feature request related to a problem?

希望可以推出gc9d01的驱动库

### Describe the solution you'd like

希望可以推出gc9d01的驱动库

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

espzav, 2025-01-28:
Hi @NingZiXi, I see that your GC9D01 driver was released in component manager https://components.espressif.com/components/ningzixi/lcd_gc9d01/versions/1.0.0
I am not sure, where is the issue.

espzav, 2025-09-10:
Closing this issue, due to inactivity. Feel free to reopen it.

closed 2025-09-10 by espzav