espressif/esp-bsp #485: Invalid value in i2c device configuration for CST816 touch driver

0xPIT, 2025-01-16:
The configuration in the header file for the CST816 driver causes a runtime assertion when used with `i2c_master_bus_add_device()`:

`E (715) i2c.master: i2c_master_bus_add_device(990): invalid scl frequency`

This is because `i2c_master_bus_add_device()` asserts `scl_speed_hz > 0`.

Please fix in this (and potentially other) drivers.

closed 2025-01-17 by tore-espressif

jpw168, 2025-01-17:
> [@0xPIT](https://github.com/0xPIT) Thank you for the report! This must have slipped in during our code reviews, will fix

After modification, in the file :  "components/esp_lcd/i2c/esp_lcd_panel_io_i2c_v1.c" ,   Line 60 an error occurred.

ESP_GOTO_ON_FALSE(io_config->scl_speed_hz == 0, ESP_ERR_INVALID_ARG, err, TAG, "scl_speed_hz is not need to set in legacy i2c_lcd driver");

Simply making changes has poor compatibility.

tore-espressif, 2025-01-17:
@jpw168 Please se it to 0, in case you still use the legacy I2C driver