espressif/esp-bsp #489: Esp32P4-mipi-dsi example code not working

Vineetha1705, 2025-01-20:
### Board

Esp32-P4-Function-EV-Board

### Hardware Description

MIPI Display attached to Esp32P4-Function-EV-Board.

### IDE Name

Espressif IDE(V5.3.2)

### Operating System

Windows11

### Description

I am working with an ESP32-P4 Function-EV-Board and using the Espressif IDE (v5.3.2). I imported the MIPI-DSI example and completed the compilation. After flashing the code, no data is displayed on the MIPI display. However, the output on Tera Term is :

SP-ROM:esp32p4-eco2-20240710
Build:Jul 10 2024
rst:0x3 (SW_SYS_RESET),boot:0xc (SPI_FAST_FLASH_BOOT)
Core0 Saved PC:0x4ff00766
Core1 Saved PC:0x4ff03afa
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x18e4
load:0x4ff2abd0,len:0xd74
load:0x4ff2cbd0,len:0x322c
entry 0x4ff2abda
I (203) esp_image: segment 5: paddr=00085318 vaddr=4ff0a7m
I (240) hex_psram: dev id : 0x03 (generation 4)
I (245) hex_psram: density : 0x07 (256 Mbit)
I (251) hex_psram: good-die : 0x06 (Pass)
I (256) hex_psram: SRF : 0x02 (Slow Refresh)
I (262) hex_psram: BurstType : 0x00 ( Wrap)
I (267) hex_psram: BurstLen : 0x03 (2048 Byte)
I (272) hex_psram: BitMode : 0x01 (X16 Mode)
I (277) hex_psram: Readlatency : 0x04 (14 cycles@Fixed)
I (283) hex_psram: DriveStrength: 0x00 (1/1)
I (289) MSPI DQS: tuning success, best phase id is 2
I (472) MSPI DQS: tuning success, best delayline id is 11
I (473) esp_psram: Found 32MB PSRAM device
I (473) esp_psram: Speed: 200MHz
I (476) hex_psram: psram CS IO is dedicated
I (480) cpu_start: Multicore app
I (1434) esp_psram: SPI SRAM memory test OK
W (1443) clk: esp_perip_clk_init() has not been implemented yet
I (1450) cpu_start: Pro cpu start user code
I (1450) cpu_start: cpu freq: 360000000 Hz
I (1450) app_init: Application information:
I (1453) app_init: Project name: mipi_dsi_panel
I (1459) app_init: App version: 1
I (1463) app_init: Compile time: Jan 18 2025 11:52:37
I (1469) app_init: ELF file SHA256: 5c42771e6...
I (1475) app_init: ESP-IDF: v5.3.2-dirty
I (1480) efuse_init: Min chip rev: v0.1
I (1485) efuse_init: Max chip rev: v1.99
I (1490) efuse_init: Chip rev: v1.0
I (1495) heap_init: Initializing. RAM available for dynamic allocation:
I (1502) heap_init: At 4FF24A40 len 00016580 (89 KiB): RAM
I (1508) heap_init: At 4FF3AFC0 len 00004BF0 (18 KiB): RAM
I (1514) heap_init: At 4FF40000 len 00060000 (384 KiB): RAM
I (1521) heap_init: At 50108080 len 00007F80 (31 KiB): RTCRAM
I (1527) heap_init: At 3010005C len 00001FA4 (7 KiB): TCM
I (1533) esp_psram: Adding pool of 32768K of PSRAM memory to heap allocator
I (1542) spi_flash: detected chip: generic
I (1545) spi_flash: flash io: dio
W (1549) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (1563) main_task: Started on CPU0
[... 343 lines cut ...]
    };
    esp_timer_handle_t lvgl_tick_timer = NULL;
    ESP_ERROR_CHECK(esp_timer_create(&lvgl_tick_timer_args, &lvgl_tick_timer));
    ESP_ERROR_CHECK(esp_timer_start_periodic(lvgl_tick_timer, EXAMPLE_LVGL_TICK_PERIOD_MS * 1000));

    // LVGL APIs are meant to be called across the threads without protection, so we use a mutex here
    lvgl_api_mux = xSemaphoreCreateRecursiveMutex();
    assert(lvgl_api_mux);

    ESP_LOGI(TAG, "Create LVGL task");
    xTaskCreate(example_lvgl_port_task, "LVGL", EXAMPLE_LVGL_TASK_STACK_SIZE, NULL, EXAMPLE_LVGL_TASK_PRIORITY, NULL);

    ESP_LOGI(TAG, "Display LVGL Meter Widget");
    // Lock the mutex due to the LVGL APIs are not thread-safe
    if (example_lvgl_lock(-1)) {
        example_lvgl_demo_ui(display);
        // Release the mutex
        example_lvgl_unlock();
    }
}
```

### I have checked existing issues, README.md and ESP32 Forum

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

espzav, 2025-01-20:
Hi @Vineetha1705, why did not use BSP and `LVGL port`?

suda-morris, 2025-02-18:
Have you connected the LCD and the development board correctly? https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32p4/esp32-p4-function-ev-board/user_guide.html#hardware-setup

ziyaboraEsmer, 2025-07-07:
solution is beyond stupid, the pins for back light and lcd reset are set up wrong in the example, under main/mipi_ds_example_main.c in lines 60 and 61, change the values from -1, -1 to 26,27 respectively and code works. I tried everything then coincidentaly found this as the solution when comparing pin numbers with a different working example.