espressif/esp-bsp #491: ESP32_S3_EYE BSP memory issue after calling bsp_display_lock(0)

100312dog, 2025-01-22:
### Board

ESP32-S3-EYE

### Hardware Description

Official esp32-s3-eye

### IDE Name

Vscode

### Operating System

ubuntu22

### Description

When enable CONFIG_BSP_DISPLAY_LVGL_BUFFER_IN_PSRAM option in sdkconfig
After the fisrt time i run ```bsp_display_lock(0); ```
100K internal ram is consumed.
I think the memory consumption here is abnormal.
Test code:
[bsp_display_lock_test.zip](https://github.com/user-attachments/files/18504631/bsp_display_lock_test.zip)
terminal output
[image]

### Sketch

```cpp
#include "bsp/esp-bsp.h"

extern "C" void app_main(void)
{
    bsp_display_start();
    ESP_ERROR_CHECK(bsp_display_backlight_on());

    int psram = heap_caps_get_free_size(MALLOC_CAP_SPIRAM);
    int internal = heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
    printf("%d, %d\n", psram, internal);

    bsp_display_lock(0);
    psram = heap_caps_get_free_size(MALLOC_CAP_SPIRAM);
    internal = heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
    printf("%d, %d\n", psram, internal);

    bsp_display_unlock();

    bsp_display_lock(0);
    bsp_display_unlock();
    psram = heap_caps_get_free_size(MALLOC_CAP_SPIRAM);
    internal = heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
    printf("%d, %d\n", psram, internal);
}

# This file was generated using idf.py save-defconfig. It can be edited manually.
# Espressif IoT Development Framework (ESP-IDF) 5.5.0 Project Minimal Configuration
#
CONFIG_IDF_TARGET="esp32s3"
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_COMPILER_OPTIMIZATION_PERF=y
CONFIG_SPIRAM=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_XIP_FROM_PSRAM=y
CONFIG_SPIRAM_SPEED_80M=y
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240=y
CONFIG_ESP32S3_INSTRUCTION_CACHE_32KB=y
CONFIG_ESP32S3_DATA_CACHE_64KB=y
CONFIG_ESP32S3_DATA_CACHE_LINE_64B=y
CONFIG_ESP_SYSTEM_ALLOW_RTC_FAST_MEM_AS_HEAP=n
CONFIG_FATFS_LFN_HEAP=y
CONFIG_FREERTOS_HZ=1000
CONFIG_FREERTOS_VTASKLIST_INCLUDE_COREID=y
CONFIG_FREERTOS_GENERATE_RUN_TIME_STATS=y
CONFIG_BSP_SD_FORMAT_ON_MOUNT_FAIL=y
```

### I have checked existing issues, README.md and ESP32 Forum

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

tore-espressif, 2025-01-23:
@100312dog thank you for the  report.

If you place the LVGL buffers to SPIRAM, they will be dynamically allocated from LVGL task. Since ESP32-S3 is dual-CPU, both the main task and LVGL task can run at the same time.

In your sketch, in between `heap_caps_get_free_size()` calls, the LVGL tasks allocates the buffers and you see less RAM

Eg if you add a delay after `bsp_display_start` you will different results
``` c
bsp_display_start();
    vTaskDelay(10);
```

Here is heap trace log with backtrace for your convenience

<details>
  <summary>Heap trace</summary>

```
  ====== Heap Trace: 4 records (10 capacity) ======
 32768 bytes (@ 0x3fca5a58, Internal) allocated CPU 1 ccount 0x03da4454 caller 0x4037a503:0x4037e180:0x42042b1b:0x4200c69f:0x4200ba59:0x4200be8e:0x4200c278:0x4200b9ca:0x4202c8f3:0x4202ccea
--- 0x4037a503: heap_caps_aligned_alloc at C:/esp_repos/esp-idf/master/components/heap/heap_caps.c:535
0x4037e180: setup_priv_desc at C:/esp_repos/esp-idf/master/components/esp_driver_spi/src/gpspi/spi_master.c:1155
0x42042b1b: spi_device_queue_trans at C:/esp_repos/esp-idf/master/components/esp_driver_spi/src/gpspi/spi_master.c:1212
0x4200c69f: panel_io_spi_tx_color at C:/esp_repos/esp-idf/master/components/esp_lcd/spi/esp_lcd_panel_io_spi.c:394
0x4200ba59: esp_lcd_panel_io_tx_color at C:/esp_repos/esp-idf/master/components/esp_lcd/src/esp_lcd_panel_io.c:29
0x4200be8e: panel_st7789_draw_bitmap at C:/esp_repos/esp-idf/master/components/esp_lcd/src/esp_lcd_panel_st7789.c:227
0x4200c278: esp_lcd_panel_draw_bitmap at C:/esp_repos/esp-idf/master/components/esp_lcd/src/esp_lcd_panel_ops.c:37
0x4200b9ca: lvgl_port_flush_callback at C:/github_repos/esp-bsp/components/esp_lvgl_port/src/lvgl9/esp_lvgl_port_disp.c:624
0x4202c8f3: call_flush_cb at C:/github_repos/esp-bsp/examples/display_camera/managed_components/lvgl__lvgl/src/core/lv_refr.c:1212
0x4202ccea: draw_buf_flush at C:/github_repos/esp-bsp/examples/display_camera/managed_components/lvgl__lvgl/src/core/lv_refr.c:1181

 32768 bytes (@ 0x3fcada5c, Internal) allocated CPU 1 ccount 0x03dc09e0 caller 0x4037a503:0x4037e180:0x42042b1b:0x4200c69f:0x4200ba59:0x4200be8e:0x4200c278:0x4200b9ca:0x4202c8f3:0x4202ccea
--- 0x4037a503: heap_caps_aligned_alloc at C:/esp_repos/esp-idf/master/components/heap/heap_caps.c:535
0x4037e180: setup_priv_desc at C:/esp_repos/esp-idf/master/components/esp_driver_spi/src/gpspi/spi_master.c:1155
0x42042b1b: spi_device_queue_trans at C:/esp_repos/esp-idf/master/components/esp_driver_spi/src/gpspi/spi_master.c:1212
0x4200c69f: panel_io_spi_tx_color at C:/esp_repos/esp-idf/master/components/esp_lcd/spi/esp_lcd_panel_io_spi.c:394
0x4200ba59: esp_lcd_panel_io_tx_color at C:/esp_repos/esp-idf/master/components/esp_lcd/src/esp_lcd_panel_io.c:29
0x4200be8e: panel_st7789_draw_bitmap at C:/esp_repos/esp-idf/master/components/esp_lcd/src/esp_lcd_panel_st7789.c:227
0x4200c278: esp_lcd_panel_draw_bitmap at C:/esp_repos/esp-idf/master/components/esp_lcd/src/esp_lcd_panel_ops.c:37
0x4200b9ca: lvgl_port_flush_callback at C:/github_repos/esp-bsp/components/esp_lvgl_port/src/lvgl9/esp_lvgl_port_disp.c:624
0x4202c8f3: call_flush_cb at C:/github_repos/esp-bsp/examples/display_camera/managed_components/lvgl__lvgl/src/core/lv_refr.c:1212
0x4202ccea: draw_buf_flush at C:/github_repos/esp-bsp/examples/display_camera/managed_components/lvgl__lvgl/src/core/lv_refr.c:1181

 32768 bytes (@ 0x3fcb5a60, Internal) allocated CPU 1 ccount 0x03ddc834 caller 0x4037a503:0x4037e180:0x42042b1b:0x4200c69f:0x4200ba59:0x4200be8e:0x4200c278:0x4200b9ca:0x4202c8f3:0x4202ccea
--- 0x4037a503: heap_caps_aligned_alloc at C:/esp_repos/esp-idf/master/components/heap/heap_caps.c:535
0x4037e180: setup_priv_desc at C:/esp_repos/esp-idf/master/components/esp_driver_spi/src/gpspi/spi_master.c:1155
0x42042b1b: spi_device_queue_trans at C:/esp_repos/esp-idf/master/components/esp_driver_spi/src/gpspi/spi_master.c:1212
0x4200c69f: panel_io_spi_tx_color at C:/esp_repos/esp-idf/master/components/esp_lcd/spi/esp_lcd_panel_io_spi.c:394
0x4200ba59: esp_lcd_panel_io_tx_color at C:/esp_repos/esp-idf/master/components/esp_lcd/src/esp_lcd_panel_io.c:29
0x4200be8e: panel_st7789_draw_bitmap at C:/esp_repos/esp-idf/master/components/esp_lcd/src/esp_lcd_panel_st7789.c:227
0x4200c278: esp_lcd_panel_draw_bitmap at C:/esp_repos/esp-idf/master/components/esp_lcd/src/esp_lcd_panel_ops.c:37
0x4200b9ca: lvgl_port_flush_callback at C:/github_repos/esp-bsp/components/esp_lvgl_port/src/lvgl9/esp_lvgl_port_disp.c:624
0x4202c8f3: call_flush_cb at C:/github_repos/esp-bsp/examples/display_camera/managed_components/lvgl__lvgl/src/core/lv_refr.c:1212
0x4202ccea: draw_buf_flush at C:/github_repos/esp-bsp/examples/display_camera/managed_components/lvgl__lvgl/src/core/lv_refr.c:1181

 16896 bytes (@ 0x3fcbda64, Internal) allocated CPU 1 ccount 0x03decd78 caller 0x4037a503:0x4037e180:0x42042b1b:0x4200c69f:0x4200ba59:0x4200be8e:0x4200c278:0x4200b9ca:0x4202c8f3:0x4202ccea
--- 0x4037a503: heap_caps_aligned_alloc at C:/esp_repos/esp-idf/master/components/heap/heap_caps.c:535
0x4037e180: setup_priv_desc at C:/esp_repos/esp-idf/master/components/esp_driver_spi/src/gpspi/spi_master.c:1155
0x42042b1b: spi_device_queue_trans at C:/esp_repos/esp-idf/master/components/esp_driver_spi/src/gpspi/spi_master.c:1212
0x4200c69f: panel_io_spi_tx_color at C:/esp_repos/esp-idf/master/components/esp_lcd/spi/esp_lcd_panel_io_spi.c:394
[... cut ...]
0x4202ccea: draw_buf_flush at C:/github_repos/esp-bsp/examples/display_camera/managed_components/lvgl__lvgl/src/core/lv_refr.c:1181

====== Heap Trace Summary ======
Mode: Heap Trace Leaks
115200 bytes 'leaked' in trace (4 allocations)
records: 4 (10 capacity, 4 high water mark)
total allocations: 14
total frees: 10
================================
```
</details>

It means that we hit this condition, because ESP32-S3 cannot DMA access SPIRAM

https://github.com/espressif/esp-idf/blob/d30e4d9960afd0d1ed067fef741126af810f675e/components/esp_driver_spi/src/gpspi/spi_master.c#L1151

100312dog, 2025-01-24:
@tore-espressif  Thank you for your explanation, i understand what's going on here. I'll close the issue.

closed 2025-01-24 by 100312dog

tore-espressif, 2025-01-24:
I double checked in esp-idf and ESP32-S3 _can_ access SPIRAM through DMA, will test and update the code if it works

100312dog, 2025-01-24:
@tore-espressif It would be great if it works, the internal memory on s3 is really tight.

tore-espressif, 2025-01-24:
I tested on HW, and the image was broken. So we need the intermediate buffer in internal RAM :(