espressif/esp-bsp #501: [esp32-p4] esp_lcd_touch_gt911_read_data false positive release reports

chegewara, 2025-02-03:
Hi team,
there is small issue with false release report with lvgl on esp32-p4 (and probably all boards with GT911).
It is because driver does not follow programming guide
```
[0x814E]:
Bit7: Buffer status, 1 = coordinate (or key) is ready for host to read; 0 = coordinate (or key) is not ready and
data is not valid. After reading coordinates, host should configure this flag (or the entire byte) to 0 via I2C.
```

This small change seems to work, but you may find a better solution (maybe return something else than ESP_OK instead)
https://github.com/espressif/esp-bsp/blob/7969b01008bf4097f3386e7d175ab5dbddbc9a71/components/lcd_touch/esp_lcd_touch_gt911/esp_lcd_touch_gt911.c#L209-L225
```
    uint8_t repeat = 0;

    assert(tp != NULL);
repeat:
    err = touch_gt911_i2c_read(tp, ESP_LCD_TOUCH_GT911_READ_XY_REG, buf, 1);
    ESP_RETURN_ON_ERROR(err, TAG, "I2C read error!");

    /* Any touch data? */
    if ((buf[0] & 0x80) == 0x00) {
        // touch_gt911_i2c_write(tp, ESP_LCD_TOUCH_GT911_READ_XY_REG, clear);       <---- dont clear buffer if data has not been read and processed

    esp_rom_delay_us(3000);    <----- with shorter delay it may take up to 10 loops, so 3ms seems to be optimal
    if(repeat++ < 3)
        goto repeat;
```

I am using v1.1.0

Thanks

espzav, 2025-03-18:
Hi, I checked this issue. There shouldn't be delay, because the upper layer (caller) is responsible for calling this read in loop with delay and sleep the task. 
And we don't read data if data are not ready. I am not sure, if the clear is ok or not now.

chegewara, 2025-03-18:
Ok, i will maintain own version then.
Like I said, library is reporting false positive release events (read 0 touch) when it can't read from g911.
Maybe the code should report -1 or something, but for me it was the quickest solution.

Thanks

federico-carbone, 2025-11-09:
Hello,
I'm using a custom board with a UI built with slint. When I press the touch
without releasing it, the UI keeps triggering press and release events.
Basically slint runs this on every render cycle (which can, but not only,
be triggered by a touch interrupt):

```
esp_lcd_touch_read_data();
bool touchpad_pressed = esp_lcd_touch_get_coordinates(...);
```
[reference](https://github.com/slint-ui/slint/blob/master/api%2Fcpp%2Fesp-idf%2Fslint%2Fsrc%2Fslint-esp.cpp#L191-L195)

In case of consecutive cycles, like for animations, the second call to
`esp_lcd_touch_read_data` reads no data from gt911, and second call to
`esp_lcd_touch_get_coordinates` returns `false` since the previous call to
this function invalidates data.

Now, in my opinion, there are 2 solutions:
- the one I proposed, which consists of not invalidating data
- the one OP proposed, which consists in returning a different return value
from `esp_lcd_touch_read_data` in case data is not ready to be read (for me
not ideal since I would anyway need to fix slint too)