espressif/esp-bsp #533: esp_io_expander 超时时间过短导致经常看门狗复位

johnzhy, 2025-03-05:
### Board

ESP32-S3

### Hardware Description

ESP32S3 I2C驱动PCA9555

### IDE Name

VSCODE ESP-IDFV5.3.2

### Operating System

Windows11

### Description

默认看门狗时间300ms，使用pca9555驱动led闪烁，不定期触发看门狗复位，看门狗时间调长到3000ms也没用，
后来发现是因为I2C_TIMEOUT_MS 默认为10，太短了，因为idf中默认的freertos节拍是100，也就是一个节拍10ms，
可能会导致实际超时时间远小于10ms，将超时时间设置为100，没有再出现ic2触发看门狗复位

### Sketch

```cpp
static void i2c_bus_init(void)
{
    const i2c_config_t i2c_conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = I2C_MASTER_SDA_IO,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_io_num = I2C_MASTER_SCL_IO,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = 100000};
    ESP_ERROR_CHECK(i2c_param_config(I2C_MASTER_NUM, &i2c_conf));
    ESP_ERROR_CHECK(i2c_driver_install(I2C_MASTER_NUM, i2c_conf.mode, 0, 0, 0));
}

static void i2c_dev_tca95xx_16bit_init(void)
{
    ESP_ERROR_CHECK(esp_io_expander_new_i2c_tca95xx_16bit(I2C_MASTER_NUM, I2C_ADDRESS, &io_expander));
}
void run_led(uint8_t on)
{
    esp_err_t ret = esp_io_expander_set_level(io_expander, RUN_LED_PIN, on ^ 1);
    if (ret != ESP_OK)
    {
        ESP_LOGE(TAG, "set run led level failed");
    }
}
```

### I have checked existing issues, README.md and ESP32 Forum

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

espzav, 2025-03-18:
@Lzw655 Please, could you help with this issue?

Lzw655, 2025-03-19:
Hi @johnzhy, 你能否尝试将 freertos tick 频率调高至 1000？通常我们都是采用这个频率来提高性能。

johnzhy, 2025-03-19:
我可以调整，但是ESPIDF默认频率是100，那岂不是在默认配置下组件不能正常工作，或者这个超时时间改成可配置的。

Lzw655, 2025-03-19:
这确实会导致问题，我们后续也将通过新增 Kconfig 配置项实现可配置的超时时间，感谢你的反馈与建议！

johnzhy, 2025-03-19:
感谢回复，那我关闭issue了

closed 2025-03-19 by johnzhy

Lzw655, 2025-03-19:
@johnzhy 我刚才看了一下源代码，你现在使用的组件版本应该是 `1.x` 吧，最新的 `2.x` 组件使用了 ESP-IDF 中新的 I2C 驱动并且配置超时时间为 1000ms。实在抱歉，`1.x` 版本后续应该无法再继续更新了。

johnzhy, 2025-03-19:
哦哦，明白，因为有组件用的是旧版i2c驱动，所以用的1.x版本，那就这样吧，要不改超时，要不改rtos频率