espressif/esp-bsp #544: lvgl.h found in component lvgl__lvgl which is already in the requirements list of espressif__esp_lvgl_port

Tan-fenyuan, 2025-03-17:
### Board

espressif__esp_lvgl_port

### Hardware Description

ST77916

### IDE Name

espIdf

### Operating System

window11

### Description

[image]

lvgl.h found in component lvgl__lvgl which is already in the requirements list of espressif__esp_lvgl_port

### Sketch

```cpp
In file included from /home/tanfenyuan/MilkBot/components/St77916/src/LvglINIT.cpp:10:
/home/tanfenyuan/MilkBot/managed_components/espressif__esp_lvgl_port/include/esp_lvgl_port.h:15:10: fatal error: lvgl.h: No such file or directory
   15 | #include "lvgl.h"
      |          ^~~~~~~~
```

### I have checked existing issues, README.md and ESP32 Forum

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

espzav, 2025-04-02:
Hi, please could you provide more info about your project (which components do you use or did you try any example)? Th ecomponent `esp_lvgl_port` is working properly - no compilation errors. We are using it in many examples in BSP. Please try some example: https://github.com/espressif/esp-bsp/tree/master/examples

tore-espressif, 2025-04-07:
@Tan-fenyuan In case you did not specifiy dependencies correctly in your CMakeLists.txt you might miss this compile definition 

https://github.com/lvgl/lvgl/blob/ac8639ffef8687e333f64478d9a4796fd3322830/env_support/cmake/esp.cmake#L67

You can add it manually to your project and see if it fixes your problem

tore-espressif, 2025-09-17:
Closing due to inactivity. Feel free to reopne if the problem perists

closed 2025-09-17 by tore-espressif