espressif/esp-bsp #565: New Driver Component : AHT20 I2C driver with thread safe calls from idf>=5.4.

jeetrohan, 2025-04-21:
### Related area

Driver component

### Hardware specification

additional support

### Is your feature request related to a problem?

Current AHT20 driver (if my understanding of it is not wrong) doesn't uses the thread safe I2C APIs provided in idf>=5.4.

### Describe the solution you'd like

I want to contribute an AHT20 (humidity-temperature sensor) driver that uses thread safe I2C APIs introduced in esp-idf>=5.4.

My contribution
- is (to at least an acceptable extent) documented
- is tested with latest revision of AHT20 using esp32
- is hardware agnostic (I believe as long as APIs are consistent  and I2C supported it should work on any esp-platform)
- consists a unit test
- and also have a simple demo example included to read and display the sensor on terminal.

Further, my proposed contribution has passed all the pre-commit hook tests for this repository, except copyright (which can be updated). 

### Describe alternatives you've considered

I could've contributed it esp-iot-solution, but .... I request to... lets not discuss about that 😅.

At the very least it could be released as v2.0 for existing for the existing one, because of a simple helpful feature included in this new driver(not described here).
And I have a few other components too, all tested and documented, that I'd like to contribute to this repository.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

espzav, 2025-04-24:
Hello @jeetrohan, I see that you have this component in ESP Registry - it is right place. There is official Espressif's component too, but it is not from BSP repository - https://github.com/espressif/esp-iot-solution/tree/master/components/sensors/humiture/aht20.
Inside BSP repository, we have mainly components for modules and sensors which are assembled on official Espressif's or M5Stack boards. It is very hard for us support other components - we cannot have all HW for runtime testing.

jeetrohan, 2025-04-28:
@espzav thankyou, for pointing in that direction. I have made a PR in the corresponding repository. 

I was unwilling to do it first, because of Sensor Hub integration (that thing has potential, but..... 🤐 ). 

This component has been tested on hardware, using AHT20 with esp32 and esp32 s3.

If you are also willing to accept it, I'll make a PR.

And..
I remember (may be not correctly) there used to be a DHT11 driver in this repo, but it had some issues as it was using some internal APIs which were deprecated. 
I also have copy of working DHT11 driver, that I've tested and updated to be independent of those APIs.
I would be happy to contribute that to this repo.

Also, I've updated the BH1750 driver (part of this repository) in a similar way, though it remains untested on hardware. 
If it is acceptable, I'd like to contribute that too to this repository.

Thanks again