espressif/esp-bsp #566: esp_lcd_touch_cst816s componet I2C driver conflict

centurion2021, 2025-04-24:
### Related area

esp_lcd_touch_cst816s,ESP-IDF 5.4

### Hardware specification

Custom ESP32-S3 Board, Similar to ESP32_S3_KORVO2

### Is your feature request related to a problem?

background:
I have a Custom ESP32-S3 Board to build a Voice Assistant based on ESP-ADF master version, and there is I2C confilct when I add LCD touch. When I try to use new esp_lcd_touch_cst816s v1.0.6 with new I2C driver independently, I got same error. 

1. Custom ESP32-S3 Board(ESP32_S3_KORVO2 like)
2. ESP-IDF 5.4
3. Work great with esp_lcd_touch_cst816s 1.0.3 with old I2C driver
4. Not work with esp_lcd_touch_cst816s(1.0.6) with new I2C driver

Issue:
#3 show error：
E (379) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver

### Describe the solution you'd like

How do I use new I2C driver in esp_lcd_touch_cst816s(1.0.6)？

Code with error:
void bsp_lcd_tp_init(void)
{
/**
    const i2c_config_t i2c_conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = EXAMPLE_PIN_NUM_TP_SDA,
        .sda_pullup_en = GPIO_PULLUP_DISABLE,
        .scl_io_num = EXAMPLE_PIN_NUM_TP_SCL,
        .scl_pullup_en = GPIO_PULLUP_DISABLE,
        .master.clk_speed = 400000
    };
    ESP_ERROR_CHECK(i2c_param_config(EXAMPLE_TP_PORT, &i2c_conf));
    ESP_ERROR_CHECK(i2c_driver_install(EXAMPLE_TP_PORT, i2c_conf.mode, 0, 0, 0));
**/
	i2c_master_bus_config_t i2c_mst_config = {
		.clk_source = I2C_CLK_SRC_DEFAULT,
		.i2c_port = EXAMPLE_TP_PORT,
		.scl_io_num = EXAMPLE_PIN_NUM_TP_SCL,
		.sda_io_num = EXAMPLE_PIN_NUM_TP_SDA,
		.glitch_ignore_cnt = 7,
		.flags.enable_internal_pullup = false,
	};

	i2c_master_bus_handle_t bus_handle;
	ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &bus_handle));
/****/
    esp_lcd_panel_io_handle_t io_handle = NULL;
    esp_lcd_panel_io_i2c_config_t io_config = ESP_LCD_TOUCH_IO_I2C_CST816S_CONFIG();
    esp_lcd_new_panel_io_i2c(EXAMPLE_TP_PORT, &io_config, &io_handle);

    esp_lcd_touch_config_t tp_cfg = {
        .x_max = EXAMPLE_LCD_H_RES,
        .y_max = EXAMPLE_LCD_V_RES,
        .rst_gpio_num = EXAMPLE_PIN_NUM_TP_RST,
        .int_gpio_num = EXAMPLE_PIN_NUM_TP_INT,
        .levels = {
            .reset = 0,
            .interrupt = 0,
        },
        .flags = {
            .swap_xy = 0,
            .mirror_x = 0,
            .mirror_y = 0,
[... 224 lines cut ...]
    };
    lvgl_disp = lvgl_port_add_disp(&disp_cfg);
    
    /* Add touch input (for selected screen) */
    const lvgl_port_touch_cfg_t touch_cfg = {
        .disp = lvgl_disp,
        .handle = tp_handle,
    };
    lvgl_touch_indev = lvgl_port_add_touch(&touch_cfg);

}

void app_main(void)
{
    bsp_lcd_tp_init();
    bsp_lcd_init();
    bsp_lcd_bl_init();
    
    app_lvgl();
    setup_ui(&guider_ui);
}

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

espzav, 2025-04-24:
> esp_lcd_new_panel_io_i2c(EXAMPLE_TP_PORT, &io_config, &io_handle);

Hello @centurion2021, 
this error means that you are using old and new (both) I2C drivers in your code. You can use only one of them. 
Fix:
1. I2C initialization should be like [THIS](https://github.com/espressif/esp-bsp/blob/76cc90336b34955fc76b510557b837e963b6a9e9/bsp/esp32_s3_korvo_2/esp32_s3_korvo_2.c#L139)
2. New panel `esp_lcd_new_panel_io_i2c` should be changed like [THIS](https://github.com/espressif/esp-bsp/blob/76cc90336b34955fc76b510557b837e963b6a9e9/bsp/esp32_s3_korvo_2/esp32_s3_korvo_2.c#L501)
3.

centurion2021, 2025-04-24:
> Hello [@centurion2021](https://github.com/centurion2021), this error means that you are using old and new (both) I2C drivers in your code. You can use only one of them. Fix:
>
> 1. I2C initialization [...]

Thank you, after some test I finally found what is wrong in my code. everything is work fine :)

right code should like bellow.
`void bsp_lcd_tp_init(void)
{
	i2c_master_bus_config_t i2c_mst_config = {
		.clk_source = I2C_CLK_SRC_DEFAULT,
		.i2c_port = -1,
		.scl_io_num = EXAMPLE_PIN_NUM_TP_SCL,
		.sda_io_num = EXAMPLE_PIN_NUM_TP_SDA,
		.glitch_ignore_cnt = 7,
		.flags.enable_internal_pullup = true,
	};

	i2c_master_bus_handle_t i2c_bus_handle;
	ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &i2c_bus_handle));
	
    esp_lcd_panel_io_handle_t io_handle = NULL;
    esp_lcd_panel_io_i2c_config_t tp_io_config = ESP_LCD_TOUCH_IO_I2C_CST816S_CONFIG();
    tp_io_config.scl_speed_hz = I2C_TOUCH_FREQ;
	
	ESP_ERROR_CHECK(esp_lcd_new_panel_io_i2c(i2c_bus_handle, &tp_io_config, &io_handle));

    esp_lcd_touch_config_t tp_cfg = {
        .x_max = EXAMPLE_LCD_H_RES,
        .y_max = EXAMPLE_LCD_V_RES,
        .rst_gpio_num = EXAMPLE_PIN_NUM_TP_RST,
        .int_gpio_num = EXAMPLE_PIN_NUM_TP_INT,
        .levels = {
            .reset = 0,
            .interrupt = 0,
        },
        .flags = {
            .swap_xy = 0,
            .mirror_x = 0,
            .mirror_y = 0,
        },
        .interrupt_callback = NULL,
    };

    esp_lcd_touch_new_i2c_cst816s(io_handle, &tp_cfg, &tp_handle);
}`

closed 2025-04-24 by centurion2021