espressif/esp-bsp #574: esp_lvgl_port i2c_oled example with 64x128 SH1107 rotation broken

mutatrum, 2025-05-20:
### Board

esp32s3

### Hardware Description

ESP32-S3-WROOM-1 with a 1.3 inch SH1107 64x128 LCD vertical module (f.e. [link](https://www.aliexpress.com/item/1005007555150608.html)):

[image]

### IDE Name

vscode

### Operating System

Linux

### Description

I would like to use this display in horizontal orientation, but am unable to rotate the display.

The `i2c_oled` example works out of the box:

[image]

If I change [L151](https://github.com/espressif/esp-bsp/blob/5705ac7871989df3b9a84ece3ff8a44e0b5c1d63/components/esp_lvgl_port/examples/i2c_oled/main/i2c_oled_example_main.c#L149) to `LV_DISPLAY_ROTATION_90`, I get this:

https://github.com/user-attachments/assets/95946cb8-010d-4754-b60e-c77d8a28650b

I've also tested with the `swap_xy`, `mirror_x` and `mirror_y` flags, but that gives different distortions.

### Sketch

```cpp
Sketch is an unlatered https://github.com/espressif/esp-bsp/blob/master/components/esp_lvgl_port/examples/i2c_oled/main/i2c_oled_example_main.c, except the changed LV rotation.
```

### I have checked existing issues, README.md and ESP32 Forum

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mutatrum, 2025-05-22:
Addition: it's also broken on the SSD1306, with `LV_DISPLAY_ROTATION_180`, so it seems something in the underlying code.

With `sw_rotate`  true as well as false, the screen is corrupted.

tore-espressif, 2025-05-29:
Hi @mutatrum , the bug should be fixed with upcoming LVGL release v9.3

mutatrum, 2025-05-30:
Thank you, is there an issue I can follow?

tore-espressif, 2025-05-31:
> Thank you, is there an issue I can follow?

Tracked here https://github.com/lvgl/lvgl/issues/7797 :)

mutatrum, 2025-06-10:
Fixed with LVGL 9.3. Thank you!

closed 2025-06-10 by mutatrum