espressif/esp-bsp #651: Update esp32_p4_function_ev_board sdconfig in examples.

shadiramadan, 2025-09-26:
### Board

ESP32 P4 Function EV Board

### Hardware Description

Official Eval Board w/ provided LCD.

Function EV Board v1.5.2
HMI SubBoard v1.2

### IDE Name

VSCode + ESP-IDF v5.5.1

### Operating System

Ubuntu 24.04.3 LTS

### Description

Hello!  I just picked up a P4 eval board.

I was able to run the factory installed demo without issue ✅ 
I've also been able to flash the board support package examples using ESP Launchpad. ✅ 
https://components.espressif.com/components/espressif/esp32_p4_function_ev_board/versions/5.0.1/examples?language=en

I've been able to write/build/flash my own simple LED Blink program code running in a simple RTOS task (ESP-EDF v5.5.1) ✅   ESP-IDF monitor works fine etc.

Side note- I've only been able to program via USB UART (OpenOCD + JTAG version / configuration seems broken).

What I have not been able to do is *run* any of the examples successfully after building from source and flashing. ❌ 

For example for the esp-bsp display example-

I created a simple project configuration file:

```json
{
  "esp32_p4_function_ev_board": {
    "build": {
      "compileArgs": [],
      "ninjaArgs": [],
      "buildDirectoryPath": "",
      "sdkconfigDefaults": [
        "sdkconfig.bsp.esp32_p4_function_ev_board"
      ],
      "sdkconfigFilePath": ""
    },
    "env": {},
    "flashBaudRate": "",
    "idfTarget": "",
    "monitorBaudRate": "",
    "openOCD": {
      "debugLevel": -1,
      "configs": [],
      "args": []
    },
    "tasks": {
      "preBuild": "",
      "preFlash": "",
      "postBuild": "",
      "postFlash": ""
    }
  }
}
```

After I build / flash / and open up IDF Monitor after resetting:

```
ESP-ROM:esp32p4-eco2-20240710
Build:Jul 10 2024
rst:0xc (SW_CPU_RESET),boot:0x30d (SPI_FAST_FLASH_BOOT)
Core0 Saved PC:0x4ff0325a
--- 0x4ff0325a: cpu_utility_ll_reset_cpu at /home/shadi/esp/master/esp-idf/components/hal/esp32p4/include/hal/cpu_utility_ll.h:25
--- (inlined by) esp_cpu_reset at /home/shadi/esp/master/esp-idf/components/esp_hw_support/cpu.c:49
Core1 Saved PC:0x4ff033b6
--- 0x4ff033b6: esp_cpu_wait_for_intr at /home/shadi/esp/master/esp-idf/components/esp_hw_support/cpu.c:57
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x164c
load:0x4ff29ed0,len:0xdcc
load:0x4ff2cbd0,len:0x34b0
entry 0x4ff29eda
I (28) boot: ESP-IDF v6.0-dev-2367-g5c5eb99eab 2nd stage bootloader
I (28) boot: compile time Sep 25 2025 14:15:34
I (29) boot: Multicore bootloader
I (31) boot: chip revision: v1.3
I (33) boot: efuse block revision: v0.3
I (36) boot.esp32p4: SPI Speed      : 80MHz
I (40) boot.esp32p4: SPI Mode       : DIO
I (44) boot.esp32p4: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=40070020 size=25650h (153168) map
I (114) esp_image: segment 1: paddr=00035678 vaddr=30100000 size=00044h (    68) load
I (116) esp_image: segment 2: paddr=000356c4 vaddr=4ff00000 size=0a954h ( 43348) load
I (126) esp_image: segment 3: paddr=00040020 vaddr=40000020 size=67810h (423952) map
I (189) esp_image: segment 4: paddr=000a7838 vaddr=4ff0a954 size=03e20h ( 15904) load
I (193) esp_image: segment 5: paddr=000ab660 vaddr=4ff0e780 size=02364h (  9060) load
I (196) esp_image: segment 6: paddr=000ad9cc vaddr=50108080 size=00020h (    32) load
I (204) boot: Loaded app from partition at offset 0x10000
I (205) boot: Disabling RNG early entropy source...
I (224) cpu_start: Multicore app
I (234) cpu_start: GPIO 38 and 37 are used as console UART I/O pins
I (235) cpu_start: Pro cpu start user code
I (235) cpu_start: cpu freq: 360000000 Hz
I (237) app_init: Application information:
I (240) app_init: Project name:     display
I (244) app_init: App version:      benchmark-latest-dirty
I (250) app_init: Compile time:     Sep 25 2025 14:15:24
I (255) app_init: ELF file SHA256:  e889975e1...
I (259) app_init: ESP-IDF:          v6.0-dev-2367-g5c5eb99eab
I (264) efuse_init: Min chip rev:     v0.1
I (268) efuse_init: Max chip rev:     v1.99 
I (272) efuse_init: Chip rev:         v1.3
I (276) heap_init: Initializing. RAM available for dynamic allocation:
I (282) heap_init: At 4FF22650 len 00018970 (98 KiB): RAM
I (287) heap_init: At 4FF3AFC0 len 00004BF0 (18 KiB): RAM
I (293) heap_init: At 4FF40000 len 00060000 (384 KiB): RAM
I (298) heap_init: At 501080A0 len 00007F60 (31 KiB): RTCRAM
I (303) heap_init: At 30100044 len 00001FBC (7 KiB): TCM
I (309) spi_flash: detected chip: gd
I (311) spi_flash: flash io: dio
I (315) main_task: Started on CPU0
I (325) main_task: Calling app_main()
I (325) LVGL: Starting LVGL task
W (325) ledc: GPIO 26 is not usable, maybe conflict with others
I (325) ESP32_P4_EV: MIPI DSI PHY Powered on
I (335) ESP32_P4_EV: Install MIPI DSI LCD control panel
I (335) ESP32_P4_EV: Install EK79007 LCD control panel
I (345) ek79007: version: 1.0.3
E (345) lcd.dsi: esp_lcd_new_panel_dpi(227): no memory for frame buffer
E (355) ek79007: esp_lcd_new_panel_ek79007(86): create MIPI DPI panel failed
E (355) ESP32_P4_EV: bsp_display_new_with_handles(619): New LCD panel EK79007 failed
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x4000bfec
--- 0x4000bfec: bsp_display_lcd_init at /home/shadi/organizations/datakit/hack/esp-bsp/bsp/esp32_p4_function_ev_board/esp32_p4_function_ev_board.c:849
file: "/home/shadi/organizations/datakit/hack/esp-bsp/bsp/esp32_p4_function_ev_board/esp32_p4_function_ev_board.c" line 849
func: bsp_display_lcd_init
expression: bsp_display_new_with_handles(&cfg->hw_cfg, &disp_handles)

abort() was called at PC 0x4ff06afb on core 0
--- 0x4ff06afb: _esp_error_check_failed at /home/shadi/esp/master/esp-idf/components/esp_system/esp_err.c:49
```

It seems to me the configuration for the esp4 eval board is out of date.  I wanted to reach out because it is frustrating to not be able to successfully build and run any of the provided examples- it would make my life so much easier if I had a valid reference to work off of.

### Sketch

```cpp
Official display example.
```

### Other Steps to Reproduce

All I did was follow the VSCode Espressif setup; the docs are a bit fragmented as I also had to discover the hard way how to correct `sudo usermod -aG dialout $USER` issues...

I cloned the esp-bsp project, opened the display folder in VSCode; edited the idf_component to point to the correct board; configured the project to use the correct sdkconfig; and then built and flashed; and failed to run.

I understand this chip is still in preview but the eval board has been out for over a year.  Are you guys able to build and run these examples yourselves from a fresh install?

### I have checked existing issues, README.md and ESP32 Forum

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

TjenWellens, 2025-09-29:
I'm using a [ESP32-P4-Nano](https://www.waveshare.com/wiki/ESP32-P4-Nano-StartPage)
had a the same error, which I fixed by copying parts from [StickiNote's sdkconfig](https://github.com/0015/StickiNote/blob/main/sdkconfig) (since that example app was working on my device)
steps I did to fix it (not sure if all these are needed)
1. config spiram
2. set partitions - nope
3. flash size - nope
4. spiram speed  200 - nope
5. [enable experimental feature](https://github.com/espressif/esp-idf/issues/15831)
6. spiram speed  200 - yes!

not sure if all these steps are needed, it's just the trail I happened to take (guessing maybe just the spiram stuff)

not sure if that helps for your hardware

espzav, 2025-10-24:
Hi, for using boards with ESP32-P4 ECO5, you need to change this in `menuconfig`: CONFIG_ESP32P4_SELECTS_REV_LESS_V3=n
We do not add into examples now, because, there is lot of older boards now. We will add it into README file.

espzav, 2025-12-02:
Hi all, 
in latest IDF, there is selected ESP32-P4 ECO5 as default (V3). I am closing this issue. Feel free to reopen it.

closed 2025-12-02 by espzav